DarkFlippers/unleashed-firmware #630: bug screen update flipper-z-f7-update-unlshd-064r.tgz

Alfa16bravo, 2023-10-13:
### Describe the bug.

hello, I installed the new update flipper-z-f7-update-unlshd-064r.tgz
and since then no backlight

### Reproduction

no backlight

UrSuLa360, 2023-10-13:
Have you checked the value in
_Settings -> LCD and Notifications -> LCD Backlight_
?

xMasterX, 2023-10-13:
Lets read what build with `r` means, and think about rgb leds that can't grow up by themselfs, you have to modify hardware of your flippper

[image]

closed 2023-10-13 by xMasterX