DarkFlippers/unleashed-firmware #633: Weather station is not working

DavidIgor, 2023-10-16:
weather station app is not working.. won't pick up signal! I does look normal but never receive any signals. 
Im using it on know weather station sensor(it was working with last FW) 
.

GOTO-GOSUB, 2023-10-16:
This is interesting, I had an issue getting this to work recently (when it had worked with the original firmware) but had thought it was me / finger trouble since I only looked at it very quickly. In my case it failed to pick up a Thermopro transmitter that had worked previously.

DavidIgor, 2023-10-16:
my is Auriol but not sure the model

closed 2023-10-16 by xMasterX

xMasterX, 2023-10-16:
> @xMasterX last part of your message felt like from Google support team! (Im aware that 98% of issues are caused by users not using things properly, but come-on ... Im using that app quite frequently and it was without issues prior to latest FW). and ... so what are the chances that two different users has similar issue? btw my sensor batteries are fine!

Why it works fine on my 2 flippers with different sensors?
How to replicate this issue? I have no way

If it wasn't working with my tests i could debug it and find issue

Or you think im magician who can find issue with someones sensor and fix it remotely?

DavidIgor, 2023-10-22:
it is working now, @xMasterX  it was user error.