DarkFlippers/unleashed-firmware #636: Dummy mode Error

jeanrafaellourenco, 2023-10-20:
### Describe the bug.

When activating Dummy mode, the shortcuts for the games stopped working. It is only showing the identity of the flipper when pressing any of the shortcuts

### Reproduction

1 - activate Dummy mode
2- try to access the games using any shortcut

### Logs

```Text
The logs do not record any relevant errors. Just returns the success of each action (button pressed)

1593352 [RPC] (211) Gui Send Input SUCCESS
1593353 [RPC] (212) Gui Send Input START
1593404 [RPC] (212) Gui Send Input SUCCESS
1593473 [RPC] (213) Gui Send Input START
1593533 [RPC] (213) Gui Send Input SUCCESS
1593534 [RPC] (214) Gui Send Input START
1593600 [RPC] (214) Gui Send Input SUCCESS
1593601 [RPC] (215) Gui Send Input START
1593656 [RPC] (215) Gui Send Input SUCCESS
1593695 [RPC] (216) Gui Send Input START
1593747 [RPC] (216) Gui Send Input SUCCESS
1593748 [RPC] (217) Gui Send Input START
1593801 [RPC] (217) Gui Send Input SUCCESS
1595587 [RPC] (218) Gui Send Input START
1595639 [RPC] (218) Gui Send Input SUCCESS
1595775 [RPC] (219) Gui Send Input START
```

### Anything else?

Firmware: unlshd-065e [18-10-2023]

wosk, 2023-10-21:
You have to set it preliminary via Settings->Desktop->Dummy buttons

jeanrafaellourenco, 2023-10-21:
> You have to set it preliminary via Settings->Desktop->Dummy buttons

You're right, I should have researched more. I ended up needing to hide the flipper's powers recently and I couldn't. I thought it was some post-update bug.

Thank you very much!

closed 2023-10-21 by jeanrafaellourenco