DarkFlippers/unleashed-firmware #638: NFC Duplicator

yash95bh, 2023-10-22:
### Description of the feature you're suggesting.

THIS IS A GENUINE AND HUMBLE REQUEST TO ADD THIS FEATURE

Just like icopy xs has a feature to write duplicate cards, Flipper should also have one.
Even though RFID can be done, it'd be great to duplicate NFC cards as well.

The feature should enable the user to copy all the data, check the hex codes and write it on a blank card and is not limited to writing only to magic cards.

Ex:: Since i work in hotel, we use Onity key card system.
I read a card > save the info and the UID as well> write it on a blank onity card> the new duplicate card opens the lock as well.

UrSuLa360, 2023-11-01:
If you can read the full card with the **NFC app** then save it and
try **NFC Magic** for writing in magic cards or the **NFC app** from flipper to write in a standard card.

You can modify the data using the flipper mobile app.

[https://docs.flipper.net/nfc/read](url)
[https://docs.flipper.net/nfc/magic-cards](url)

yash95bh, 2023-11-02:
1. It doesn't read the full card and stops  at sector 15/16 and keys 30/32
2. There's no option in *NFC app *to write to a standard card.
3. The saved card doesn't emulate and open the locks as well. No matter
which card I read, it doesn't unlock anything anymore.

berkgercek, 2023-11-02:
The reason your card stops at 15/16 is because the keys in the last sector are not in the built-in dict of factory keys. This means a brute force attack or key sniffing is necessary to obtain the final key. This applies to most Mifare Classic cards and will not allow 1-step emulation, not to mention writing.

Even the iCopy will require you to sniff that extra key, or use a laptop/PC to brute force it. If you sniff the reader with the flipper it should fully unlock.

yash95bh, 2023-11-02:
Okay. Let me give that a shot.

How do I write the saved on a standard classic mifare 1K ? ( and not on the
magic card)

yash95bh, 2023-11-03:
Hi,

How do I write in a standard card from the NFC app? There is no option to
do so. It only shows to write the details on the magic card and not
otherwise.

Regards.

berkgercek, 2023-11-03:
In short you cannot. The UID field of normal cards is not writable, and as the UID forms a key component of the card you cannot duplicate arbitrary 1K classic cards onto other standard cards. Only magic cards with a writable UID.

xMasterX, 2023-11-23:
Use magic cards gen1 with NFC Magic app that can be found in Apps->NFC
There is no "blank" cards, UID is written from the factory on regular cards
Get missing keys using detect reader and flipper's mobile app with mfkey32 in it

closed 2023-11-23 by xMasterX