DarkFlippers/unleashed-firmware #650: Is the firmware dead

HTAMIMI2, 2023-11-11:
### Describe the bug.

Is the firmware dead?

### Reproduction

no updates for a month last file update was three weeks ago so whats the deal?

xMasterX, 2023-11-12:
Yes

xMasterX, 2023-11-12:
If I can't have any breaks due to my health problems, you can say anything, dead, abandoned, and other things
If you haven't seen my post in discord and telegram ill say again here, most of the work is already done locally, i'm not publishing it on github yet

assasinfil, 2023-11-21:
Is the issue dead?)

xMasterX, 2023-11-21:
> Is the issue dead?)

dev branch looks alive, but `release` branch is dead :(((
still actual issue then

xMasterX, 2023-11-26:
> Do you think it still makes sense to release even though the NFC seems to be badly broken in the upstream?
> 
> I mean they did really great job but it requires some time to deal with multiple regressions. I also believe it's a lot of work to sync with upstream easily taking into account there were more than 500 files changed.

Release won't be done while there are issues with new stack
But merge was already done in `dev` branch, and some work with plugin fixes too, in apps repo
Couple problems are still in place, when we finish with them, only then it will be possible to make the release

stremovskyy, 2023-11-28:
> @xMasterX thanks a lot for your hard work! 👍

+1

GloryToGlorzo69, 2023-12-08:
Take a look into the Discord
[https://lab.flipper.net/?url=https://unleashedflip.com/fw_extra_apps/flipper-z-f7-update-874e.tgz&channel=dev-cfw&version=874e](https://lab.flipper.net/?url=https://unleashedflip.com/fw_extra_apps/flipper-z-f7-update-874e.tgz&channel=dev-cfw&version=874e) is the newest unleashed-version

xarinatan, 2023-12-10:
I'm glad to hear there's still work done on the firmware, I didn't realize there was such a huge overhaul underway on the NFC parts- Good luck on the journey! I also became a Patreon subscriber since I've been putting off on doing that for too long, hope it helps a little.

Thanks to all involved for building this, it's sincerely a great part of the Flipper Zero ecosystem that I wouldn't want to see it do without.

xMasterX, 2023-12-17:
Looks like release came out recently - https://github.com/DarkFlippers/unleashed-firmware/releases/tag/unlshd-066
Issue seems to be not actual, time to close it now

closed 2023-12-17 by xMasterX