DarkFlippers/unleashed-firmware #661: dev branch - Keeloq is broken after 29 Nov

RekordNadoev, 2023-12-03:
### Describe the bug.

Menu => Sub-GHz => Read
Flipper does reboot with "NULL pointer dereference" after Keeloq received.
Tried with Doorhan and AN Motors.

### Reproduction

Run "Menu => Sub-GHz => Read"
Push a button on "Doorhan" or "AN Motors" fob
Flipper does reboot with "Flipper crashed and was rebooted. NULL pointer dereference"

### Target

"firmware_target": 7

### Anything else?

I've bult commits:
- "firmware_commit": "8628a2c6" - Keeloq received OK
- "firmware_commit": "159aef02" - Flipper reboots
- "firmware_commit": "ec99b70b" - Flipper reboots

RekordNadoev, 2023-12-03:
"firmware_commit": "d6fcb04a" - Fixed reboots, Doorhal looks OK, but Keeloq 64bit AN-Motors shows as Keeloq 64bit Unknown.

xMasterX, 2023-12-04:
Thanks for report, will be fixed

xMasterX, 2023-12-05:
Last Commit 10444b943e31fc03aa34e40f74e1c89760783541 should fix this issues

Try building or using dev build when it compiles, should be at number 871 or 872

RekordNadoev, 2023-12-05:
Confirmed.
KeeLoq 64bit DoorHan - OK
KeeLoq 64bit AN-Motors - OK

Thank You for your great job!