DarkFlippers/unleashed-firmware #664: Automatic Or Manually Data Restoration After NFC Emulation

Dinnerboard, 2023-12-08:
### Description of the feature you're suggesting.

Description:
As a Flipper Zero user, I've encountered an issue where the data on the emulated NFC card gets modified by the card reader after each emulation session. This behavior renders the card unusable for subsequent operations.

Details:
Current Issue: In some of the readers, after performing an NFC emulation, the card reader modifies the data on the emulated card and makes it impossible to use again.

Requested Feature: I propose the implementation of an automatic data restoration functionality that ensures the original data is rewritten after each successful NFC emulation or add a menu option that allows users to manually trigger the restoration of original data on the emulated NFC card.

Maybe more easily, make it like a "GTU" card.

liszkai, 2023-12-08:
Hi

You need a chinese Gen4 GTU magic card. It have a feutere called shadow mode. This is working just like you descrobed.

Dinnerboard, 2023-12-08:
> Hi
> 
> You need a chinese Gen4 GTU magic card. It have a feutere called shadow mode. This is working just like you descrobed.

Tank you, but I would to know whether it is possible to add this feature in the emulation of the flipper zero.

Leptopt1los, 2024-01-06:
Hi, @Dinnerboard!
This function is already included in the NFС application. if the dump was changed during emulation, a file with the same name and extension .shd is created in its folder. In this case, the menu item "Restore Data Changes" appears in the dump menu. The same item appears if the dump was changed through a companion mobile application
[image]

closed 2024-01-06 by Leptopt1los