DarkFlippers/unleashed-firmware #671: Saved NFC parse error

kobleistvan, 2023-12-17:
### Describe the bug.

Hi! I'm writing about an interesting issue I've found. This only happens to SOME NFC save files, not all. i.e. Mifare Classics don't seem to have an issue, but ISO15693-3 and NTAG/Ultralight (NTAG203) do.

### Reproduction

Step 1:
Scan some NFC tags with the latest official firmware (non unleashed). i.e. v0.96.1
Step 2:
Install Unleashed firmware
Step 3:
Try to open up the files saved previously with the official firmware.

Conclusions:
For some tag types, Unleashed doesn't support the latest version of file format. Here are some examples:

Tag file with official firmware:
`Filetype: Flipper NFC device
Version: 4
# Device type can be ISO14443-3A, ISO14443-3B, ISO14443-4A, ISO14443-4B, ISO15693-3, FeliCa, NTAG/Ultralight, Mifare Classic, Mifare DESFire, SLIX, ST25TB
Device type: NTAG/Ultralight
# UID is common for all formats
UID: 04 5A 23 D2 DB 36 80
# ISO14443-3A specific data
ATQA: 00 44
SAK: 00
# NTAG/Ultralight specific data
Data format version: 2
NTAG/Ultralight type: NTAG203
Signature: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Mifare version: 00 00 00 00 00 00 00 00
Counter 0: 0
Tearing 0: 00
Counter 1: 0
Tearing 1: 00
Counter 2: 0
Tearing 2: 00
Pages total: 42
Pages read: 42
Page 0: 04 5A 23 F5
Page 1: D2 DB 36 80
Page 2: BF 48 00 00
Page 3: E1 10 12 00
Page 4: 01 03 A0 10
Page 5: 44 03 4D 92
Page 6: 08 00 41 53
Page 7: 55 53 2F 44
Page 8: 4D 52 12 0A
Page 9: 1E 74 65 78
Page 10: 74 2F 6C 6F
Page 11: 67 69 6E 4B
Page 12: 6F 62 6C 65
Page 13: 2C 57 48 51
Page 14: 70 73 45 4F
Page 15: 58 36 50 78
Page 16: 2F 32 75 4F
Page 17: 6A 42 37 36
Page 18: 30 48 67 3D
Page 19: 3D 52 0B 09
Page 20: 74 65 78 74
Page 21: 2F 6C 61 75
Page 22: 6E 63 68 50
Page 23: 72 6F 66 69
Page 24: 6C 65 20 31
Page 25: FE 00 00 31
Page 26: 00 00 00 00
Page 27: 00 00 00 00
Page 28: 00 00 00 00
Page 29: 00 00 00 00
Page 30: 00 00 00 00
Page 31: 00 00 00 00
Page 32: 00 00 00 00
Page 33: 00 00 00 00
Page 34: 00 00 00 00
Page 35: 00 00 00 00
Page 36: 00 00 00 00
Page 37: 00 00 00 00
Page 38: 00 00 00 00
Page 39: 00 00 00 00
Page 40: 00 00 00 00
Page 41: 00 00 00 00
Failed authentication attempts: 0
`

Tag file with unleashed (065)
`Filetype: Flipper NFC device
Version: 3
# Nfc device type can be UID, Mifare Ultralight, Mifare Classic or ISO15693
Device type: NTAG203
# UID is common for all formats
UID: 04 5A 23 D2 DB 36 80
# ISO14443 specific fields
ATQA: 00 44
SAK: 00
# Mifare Ultralight specific data
Data format version: 1
Signature: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Mifare version: 00 00 00 00 00 00 00 00
Counter 0: 0
Tearing 0: 00
Counter 1: 0
Tearing 1: 00
Counter 2: 0
Tearing 2: 00
Pages total: 42
Pages read: 42
Page 0: 04 5A 23 F5
Page 1: D2 DB 36 80
Page 2: BF 48 00 00
Page 3: E1 10 12 00
Page 4: 01 03 A0 10
Page 5: 44 03 4D 92
Page 6: 08 00 41 53
Page 7: 55 53 2F 44
Page 8: 4D 52 12 0A
Page 9: 1E 74 65 78
Page 10: 74 2F 6C 6F
Page 11: 67 69 6E 4B
Page 12: 6F 62 6C 65
Page 13: 2C 57 48 51
Page 14: 70 73 45 4F
Page 15: 58 36 50 78
Page 16: 2F 32 75 4F
Page 17: 6A 42 37 36
Page 18: 30 48 67 3D
Page 19: 3D 52 0B 09
Page 20: 74 65 78 74
Page 21: 2F 6C 61 75
Page 22: 6E 63 68 50
Page 23: 72 6F 66 69
Page 24: 6C 65 20 31
Page 25: FE 00 00 31
Page 26: 00 00 00 00
Page 27: 00 00 00 00
Page 28: 00 00 00 00
Page 29: 00 00 00 00
Page 30: 00 00 00 00
Page 31: 00 00 00 00
Page 32: 00 00 00 00
Page 33: 00 00 00 00
Page 34: 00 00 00 00
Page 35: 00 00 00 00
Page 36: 00 00 00 00
Page 37: 00 00 00 00
Page 38: 00 00 00 00
Page 39: 00 00 00 00
Page 40: 00 00 00 00
Page 41: 00 00 00 00
Failed authentication attempts: 0
`

Another example:

Tag with official firmware:
`Filetype: Flipper NFC device
Version: 4
# Device type can be ISO14443-3A, ISO14443-3B, ISO14443-4A, ISO14443-4B, ISO15693-3, FeliCa, NTAG/Ultralight, Mifare Classic, Mifare DESFire, SLIX, ST25TB
Device type: ISO15693-3
# UID is common for all formats
UID: E0 16 24 66 1E D8 EF C0
# ISO15693-3 specific data
# Data Storage Format Identifier
DSFID: 02
# Application Family Identifier
AFI: 00
# IC Reference - Vendor specific meaning
IC Reference: 02
# Lock Bits
Lock DSFID: false
Lock AFI: false
# Number of memory blocks, valid range = 1..256
Block Count: 52
# Size of a single memory block, valid range = 01...20 (hex)
Block Size: 04
Data Content: F7 08 D5 9A 82 18 60 20 00 38 00 10 1C 48 33 00 1B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 2A 80 53 42 1F 90 53 42 33 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 C2 11 10 07 C0 05 1B 01 72 54 C4 6B 40 73 2E 80 10 03 C9 17 0C 00 00 00 00 00 00 3D 00 08 2C 58 00 00 00 00 00 00 00 00
# Block Security Status: 01 = locked, 00 = not locked
Security Status: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
`

Tag scanned with unleashed:
`Filetype: Flipper NFC device
Version: 3
# Nfc device type can be UID, Mifare Ultralight, Mifare Classic or ISO15693
Device type: ISO15693
# UID is common for all formats
UID: E0 16 24 66 1E D8 EF C0
# Data Storage Format Identifier
DSFID: 02
# Application Family Identifier
AFI: 00
IC Reference: 02
# Number of memory blocks, usually 0 to 256
Block Count: 52
# Size of a single memory block, usually 4
Block Size: 04
Data Content: F7 08 D5 9A 82 18 60 20 00 38 00 10 1C 48 33 00 1B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 2A 80 53 42 1F 90 53 42 33 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 C2 11 10 07 C0 05 1B 01 72 54 C4 6B 40 73 2E 80 10 03 C9 17 0C 00 00 00 00 00 00 3D 00 08 2C 58 00 00 00 00 00 00 00 00
# First byte: DSFID (0x01) / AFI (0x02) / EAS (0x04) / PPL (0x08) lock info, others: block lock info
Security Status: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
# Subtype of this card (0 = ISO15693, 1 = SLIX, 2 = SLIX-S, 3 = SLIX-L, 4 = SLIX2)
Subtype: 00
# End of ISO15693 parameters
`

As you can see, the file structure is a little bit different. I suppose this is why the NFC app cannot open some of the file types.

Suggestion:
Update the app to support also the latest format.

### Target

v065

xMasterX, 2023-12-18:
Hello, v065 will not support new version of NFC files, install 066, all works correctly at v066

closed 2023-12-18 by xMasterX

kobleistvan, 2023-12-18:
Thanks @xMasterX