DarkFlippers/unleashed-firmware #674: Crash after bluetooth connection used with  [BMI160]Air Mouse application

Z0rinSerg, 2023-12-19:
### Describe the bug.

Flipper crashed after Bluetooth connection used with  [BMI160]Air Mouse application
Sometime flipper has crashed after Bluetooth disconnecting on PC Host.
Sometime crash happened after another trying to connect by Bluetooth at the same application, when after disconnecting on PC host I try to connect to Flipper.
Crashes never happened after Bluetooth connect/disconnect with another application for example – BT Remote for Kodi. 

Flipper Zero Model: FZ.1 FW: unlshd-066e
GPIO module : BMI160 

### Reproduction

On Flipper 0:
Settings→Bluetooth→ON
Apps→GPIO→[BMI160]Air Mouse→Bluetooth
On PC Host – Bluetooth On – Control FlipperName – Connect.
All works properly, no problems.

On PC Host – Bluetooth On – Control FlipperName – Disconnect.
After that – on Flipper 0:
“Flipper crashed and was rebooted
MPU fault, possibly stack 
overflow” message appeared.

InageMantra, 2024-02-25:
I'm getting this too and it's pretty annoying.

Oddly enough it usually doesn't crash during the app itself, only after exiting Air Mouse and trying to use another app does it crash, and usually with a random error like "bus fault" or "null pointer dereferenced".

xMasterX, 2024-06-14:
Please test again on latest dev build, app was updated recently

closed 2024-09-09 by xMasterX