DarkFlippers/unleashed-firmware #684: MALVEKE Apps are not updated in the Flipper Zero IOS App

explit7, 2023-12-27:
### Describe the bug.

Malveke Apps 
https://github.com/EstebanFuentealba/MALVEKE-Flipper-Zero

are not updated in the Flipper Zero IOS App and are still on Version 1.0, while there is Version 1.1 out.

### Reproduction

Install latest unleashed Firmware
Open Flipper Zero App on IOS
Try to install one of MALVEKE Apps.
The Malveke Apps are offered lways in 1.0 Version, while Version 1.1 is out on GitHub.

### Target

Flipper Zero unleashed Firmware 0.67e

xMasterX, 2023-12-30:
So, we should ask the author of MALVEKE apps to update them in the official flipper apps catalog?
How this is related to unleashed firmware? I'm not an developer of those apps and I can't update them in the catalogue
You should ask the author

closed 2023-12-30 by xMasterX

explit7, 2023-12-30:
Yes, my mistake, i already contacted the developer of Malveke apps. Thanks for closing. Sorry for bothering!