DarkFlippers/unleashed-firmware #685: Unable to emulate classic Mifare classic 1k iso 144443-3(nfc-a)

Pmacdon15, 2023-12-29:
### Describe the bug.

I am able to read card no problem, how ever when i try to ready an emulated card from the flipper on my Arduino RFID scanner, the scanner can read the cards but not the flipper. I also tried an NFC app on my phone which reads the cards same info but dose not read the flipper. I had unleashed 66 which mentions something, I updated to 67, still have the same issue.  

### Reproduction

1. read card
2. emulate card
3. no card read

mgrybyk, 2023-12-30:
Does it work for you on 65?

Pmacdon15, 2023-12-30:
It did not, but I didn't wipe the SD card. Would that have an effect? I also tried with stock 93 and it said there was an api miss match when I tried to use the nfc app. Does updating the firmware automatically update the apps on the SD card or do I have to wipe the SD card and then update the firmware to 65 and see if it works?

This message and any attached documents are only for the use of the intended recipient(s), are confidential and may contain privileged information. Any unauthorized review, use, retransmission, or other disclosure is strictly prohibited. For information about how BVC is in compliance with the Canadian Anti-Spam Legislation (CASL), please visit our CASL Information Page or copy and paste the following URL into your web browser: https://bowvalleycollege.ca/casl. If you feel that you have received this message in error, notify the sender immediately, and delete the original message.

Pmacdon15, 2023-12-31:
its very strange to me that the issue seems to be happening on the stock firmware after 93 as well.

Pmacdon15, 2024-01-01:
Withdrawing the issue. although I can not read emulated on the Arduino, it works on my phone after I took the case off. So I do not believe it is an issue. Sorry again and thank you for your time, great work here.

Pmacdon15, 2024-01-01:
none issue.

closed 2024-01-01 by Pmacdon15