DarkFlippers/unleashed-firmware #701: Cannot write NFC v3 saved item

Airmonkdev, 2024-02-01:
### Describe the bug.

I've saved some NFC cards with fw v.065 nad opening the file I can read V3.
Now I've installed fw v.071 and if I try to write into initial card it is not working anymore. I've also tried to read with the new fw but also the v4 saved file cannot be write.
I'm I wrong? If this feature has been broken is there a plan to get it back working?

### Reproduction

1. Wrtie to initial card an old V3 saved NFC card file

Leptopt1los, 2024-02-04:
@Airmonkdev Attach that v3 dump please

Airmonkdev, 2024-02-05:
Attached!

Thanks

Leptopt1los, 2024-02-08:
@Airmonkdev I tested your dump. The problem is that he has unknown A keys to some sectors (0, 1, 2, 3, 8). Get these keys using any known attack on MF Classic, writing from a full dump works without errors

closed 2024-02-08 by Leptopt1los

Airmonkdev, 2024-02-09:
@Leptopt1los really thanks for your time.
I'm newbie sorry for the questions: why with the old Flipper's FW I was able to write even if these keys are missing?
Could you please also help me with some link or info to fix this A keys issue?

Leptopt1los, 2024-02-09:
> @Leptopt1los really thanks for your time. I'm newbie sorry for the questions: why with the old Flipper's FW I was able to write even if these keys are missing? Could you please also help me with some link or info to fix this A keys issue?

I'll explore later how it worked on pre-factor versions of the firmware. We may fix this behavior later, but it will be a low priority task

To get these keys, install 065-e version (https://github.com/DarkFlippers/unleashed-firmware/releases/tag/unlshd-065), go to apps->nfc->flipper nested and follow this guide: https://github.com/AloneLiberty/FlipperNested/wiki/Usage-guide

Airmonkdev, 2024-02-12:
@Leptopt1los thank you so much, with your help I'm now able to write back the NFC file also with the last firmware version.
I've also learned how to use nested app... great!

I think it will be good, even if with low priority task, to be able to write the files where missing keys are present like the old release did. Of course it's up to guys to decide. :)