DarkFlippers/unleashed-firmware #707: [Sub-Ghz] - Apps - playlist is not sending any RF

MarcelZwolle, 2024-02-07:
### Describe the bug.

When I try to transmit some pre-recorded Codes after each other, with the playlist, no RF comes out. I see on the screen that it goes through the list of added codes, buth no LED or RF comes out.

### Reproduction

go to: Apps - Sub-Ghz - Sub-Ghz playlist - load the file (even the example_playlist has this) In the screen I see '4 items in playlist' and set repeat to 10 times (so I can see what happens), press start and see it runs through the list very fast, but no RF comes out (and the LED isn't burning)

xMasterX, 2024-02-10:
example playlist is pointing to non existing files, you need to create your own playlist with correct paths

also there is no LED indication of transmission, that might be added in the app later

App is working correctly, tested right now

closed 2024-02-10 by xMasterX

MarcelZwolle, 2024-02-11:
[xMasterX](https://github.com/xMasterX) I'm sorry for your time, you were absolutely right! I made a typo in the path...

May I include 2 wishes? Maybe it's possibe to add something like 'missing file' when it can't be found and it should be great if you can see it's transmitting (LED)

Thanks!! Keep up the good work!