DarkFlippers/unleashed-firmware #717: Radiosonde app

DavidIgor, 2024-02-17:
### Description of the feature you're suggesting.

as Flipper is capable of receiving signal , it would be great to have ability to receive data from radiosonde (weather balloons) ...  existing weatherstation  app could be duplicated and changed  as it is fairly good interface.

daskalovst, 2024-05-07:
This feature is only usable during the final stage of recovery when you're close to the sonde. Typically, the signal from sondes is very weak, making it nearly impossible to read the data when the balloon is far from your location. However, it's useful because it eliminates the need for additional hardware. The flipper's pocket size makes it perfect for use in dense forests, especially when you already have a rough idea of the balloon's location. It can save a lot of time walking around the forest area.

DavidIgor, 2024-05-07:
absolutely, but with external cc1101 module and good antenna... it may be way more capable

derskythe, 2024-07-24:
@DavidIgor 
in my region I can't find any radisonde even on HackRF with good antenna.