DarkFlippers/unleashed-firmware #724: Flipper crashes and reboots after closing Evil Portal.

Awlexegrecki, 2024-02-28:
### Describe the bug.

 After stopping and exiting evil portal, the flipper crashes and reboots. It shows "NULL pointer dereference". Currently on Unleashed 071e. 

[image]

### Reproduction

1. Switch on
2. Navigate to GPIO Apps
3. Open Evil Portal
4. Start Evil Portal (while Official ESP32 module is plugged in with marauder + evil portal fw)
5. Stop Evil Portal
6. Close Evil Portal

vaidasnet, 2024-03-20:
Unleashed RIP

xMasterX, 2024-04-13:
Should be fixed in latest release 072 and dev branch too

closed 2024-04-13 by xMasterX