DarkFlippers/unleashed-firmware #732: Writing saved RFID tag bricks the tag

YLSnewb, 2024-03-13:
### Describe the bug.

I have previously used the RFID function in the F0 to mange keyfobs for my building.

I just received a new batch of writable keyfobs, and I was able to read them, but after attempting to write to them, they appear dead -- I'm not able to read them, or write to them.

It may be a quality issue with the fobs, but I've ordered some replacements and will report back.

### Reproduction

The process is:

Test the new keyfob by reading the existing code (successful).
Selecting the keyfob to copy from the Flipper's saved menu.
Select 'write' and apply the same fob to the back of the flipper.

The write never finishes, and the fob is broken from that point forward.

vaidasnet, 2024-03-20:
not worth reporting, Unleashed RIP

xMasterX, 2024-04-13:
We can't replicate this issue on our tags, make sure you are using proper T5577 tag, and it has no password on it before writing

closed 2024-04-13 by xMasterX