DarkFlippers/unleashed-firmware #735: SubGhz: Keeloq Cardin HCS301 protocol support

AxelCrow91, 2024-03-19:
### Description of the feature you're suggesting.

Add support to decode Cardin S449 devices

assasinfil, 2024-03-19:
Need raw captures to implement it.

AxelCrow91, 2024-03-20:
[Cardin_S449.zip](https://github.com/DarkFlippers/unleashed-firmware/files/14673184/Cardin_S449.zip)
R stand for Right Button.
L stand for Left Button.
D stand for Sinc Button.
Numbers 1 to 4 are the sequential signals transmitted.
FSK12k is the modulation set on F0 while recording the signals.

[image]

[image]

Is this enough?