DarkFlippers/unleashed-firmware #737: Authentificator crush

superyarik, 2024-03-26:
### Describe the bug.

facing authentificator crush after enter some pin or just hit center button. after crush flipper reboot. while rebooting logs disconnect

### Reproduction

open tools -> Authentificator
enter pin
press center button

### Logs

```Text
<CONNECTED>

              _.-------.._                    -,
          .-""--..,,_/ /`-,               -,  \ 
       .:"          /:/  /'\  \     ,_...,  `. |  |
      /       ,----/:/  /`\ _\~`_-"`     _;
     '      / /`"""'\ \ \.~`_-'      ,-"'/ 
    |      | |  0    | | .-'      ,/`  /
   |    ,..\ \     ,.-"`       ,/`    /
  ;    :    `/`""\`           ,/--==,/-----,
  |    `-...|        -.___-Z:_______J...---;
  :         `                           _-'
 _L_  _     ___  ___  ___  ___  ____--"`___  _     ___
| __|| |   |_ _|| _ \| _ \| __|| _ \   / __|| |   |_ _|
| _| | |__  | | |  _/|  _/| _| |   /  | (__ | |__  | |
|_|  |____||___||_|  |_|  |___||_|_\   \___||____||___|

Welcome to Flipper Zero Command Line Interface!
Read the manual: https://docs.flipper.net/development/cli
Run `help` or `?` to list available commands

Firmware version: release-cfw unlshd-072e (8b62b6cb built on 25-03-2024)

>: log
Current log level: info
Use <log ?> to list available log levels
Press CTRL+C to stop...
120778 [I][BleGap] Start: 4
126123 [I][Loader] Loading /ext/apps/Tools/totp.fap
126287 [I][Elf] Total size of loaded sections: 40160
126290 [I][Loader] Loaded in 167ms
126292 [I][TOTP APP] App version: 5.13.0
126294 [I][TOTP APP] WolfSSL version: 5.6.6
126297 [I][AnimationManager] Unload animation 'L2_Coding_in_the_shell_128x64'
126320 [I][TOTP APP] Obsolete config file version detected. Current version: 5; Actual version: 12
126349 [I][TOTP APP] Took config file backup to /ext/apps_data/totp/backups/totp.conf.20240326-6
126439 [I][TOTP APP] Applied migration to version 12
<ERROR: The device has been lost.>
<DISCONNECTED>
```

xMasterX, 2024-04-13:
Please create issue here:

https://github.com/akopachov/flipper-zero_authenticator/issues

akopachov, 2024-08-25:
Highly-likely due to too old conf file with old crypto algo. In authenticator v5 crypto algorithms v1 and v2 were by default disabled in order to have more memory for something useful. Possible solutions here are:

1. Delete conf file, which will result in loosing all the settings and tokens
2. If p1 is not acceptable, use totp artifacts from authenticator repo suffixed with `old-crypto-support`.

[Similar issue](https://github.com/akopachov/flipper-zero_authenticator/issues/209#issuecomment-1759025604)

panosmariannaki, 2024-08-25:
i fix authenticator crash if anyone have problem i connect my flipper to flipper app original app then delete entire folder apps then i reinstall latest unleashed firmware 77. via web updater, now is working like a baby no more crash authenticator and flipper. it takes 5 minutes max.

closed 2024-09-09 by xMasterX

superyarik, 2024-10-10:
@panosmariannaki now i have 78 unleashed and crashes still here

akopachov, 2024-10-10:
> @panosmariannaki now i have 78 unleashed and crashes still here

Read this:

> Highly-likely due to too old conf file with old crypto algo. In authenticator v5 crypto algorithms v1 and v2 were by default disabled in order to have more memory for something useful. Possible [...]