DarkFlippers/unleashed-firmware #744: NFC saved card can't use"Unlock with Dictionary"

wardens, 2024-04-18:
### Describe the bug.

NFC saved card can't use"Unlock with Dictionary"
I used Flipper to read a card and then sniffed the key.
Then, open saved and find the sniffing card.
choose Unlock with Dictionary.
It dosen't work.

### Reproduction

1.find a card that doesn't know the key.
2.read the card.
3.sniffing the card reader.get the key.
4.add the key to the dictionary.
5.use "Unlock with Dictionary"
6.It dosen't work.

### Target

NFC

wardens, 2024-04-18:
Because I sniffed the password, the second time I directly read it can retrieve this card

xMasterX, 2024-04-23:
What card? Unlock with dict is only for mifare classic

AztecCodes, 2024-09-25:
@wardens It really depends on which type of card it is. You can't unlock every card with the dictionary.