DarkFlippers/unleashed-firmware #747: LCD backlight not working in unleashed 071r/072r

xecaz, 2024-04-23:
### Describe the bug.

Please close this if already known, but since i installed unleashed 071r the LCD backlight is no longer working. Assumed a bug and just updated to 072r but have the same issue. Works fine on standard flipper firmware.

### Reproduction

1. unpack new flipper ;)
2. choose to update flipper to custom firmware, select the flipper-z-f7-update-unlshd-071r.tgz/flipper-z-f7-update-unlshd-072r.tgz archive. 
3. reboot.
4. version reflects that the new firmware is in place, but LCD backlight is no longer working

### Target

bug

BrianD436, 2024-04-23:
Do you have a modified flipper for RGB? Because if you have not modified it for RGB then you'll have issues. Switch to another release instead.

Leptopt1los, 2024-04-23:
closing because of @BrianD436 answer

closed 2024-04-23 by Leptopt1los

BrianD436, 2024-04-24:
> have

Going forward, read those designations. I use the Extra. It's my favorite.