DarkFlippers/unleashed-firmware #752: Sub-GHz External C1101 Module Selectability

hotdjdave, 2024-05-13:
### Describe the bug.

There used to be a setting under **Sub-GHz** > **Radio Settings** > **Module** which listed the selectable options "**Internal**" and "**External**." **External** was the way to enable an _external_ antenna (or C1101 module) and, at least to me, ensured it was using the external antenna and not the internal one. Now, going here only shows "**Internal**" and no option to select "**External**." Why was this taken away?

There should be some indication on Flipper letting you know if the internal C1101 module is being use or an external one is used.****

### Reproduction

1. Sub-GHz
2. Radio Settings
3. Module

xMasterX, 2024-05-15:
Hi, External menu item appears only when external module is connected before launching subghz app, if module was not detected there will be no External option in menu

If module is not detected that means you have bad contact or module is broken

Nothing was "taken away" or "removed" from firmware

closed 2024-05-15 by xMasterX

hotdjdave, 2024-05-16:
My new flipper arrived two days ago (May 14) and a new cc1101 module I got on eBay.  The same issue.  The "Internal" setting is the only option.  I have been using my other Flipper along with a euerodesign C1101 module for a couple of years with no issues.  I am not sure if this "auto detect" feature is the cause, but it seems to be the only variable that has changed.  NEW FLIPPER and NEW CC1101 MODULE and still no "EXTERNAL" option.