DarkFlippers/unleashed-firmware #762: Mouse Jiggler Stealth could be even more Stealthier

talktosanjay3, 2024-06-07:
### Describe the enhancement you're suggesting.

Thank you so much for all the wonderful contribution.
I was wondering if its possible to add Key strokes along with mouse movement.
Like,
1) Pressing Atl+Tab on certain intervals
2) Pressing random keys (except Enter)

xMasterX, 2024-06-09:
yes this is possible, however random keys may cause random issues :)
need to find some "safe" keys like pressing media volume up and down to get it to previous level
or something like that
since pressing some random key is not a good thing if you got some stuff running, like some console app is doing its work in terminal, you left it for a while, jiggler jiggles and randomly sends ctrl+c command to that terminal, will be not fun

talktosanjay3, 2024-06-09:
Completely agree!, It would work if the options are configurable and the user chooses what keystores he wants. 
Best Regards

dreamcasting, 2024-10-17:
Can there also be an "infinite" jiggler modification while you're in there? A max of 30 minutes is great, but I have use cases needing much longer.