DarkFlippers/unleashed-firmware #763: Sub GHZ Remote App empty

AztecCodes, 2024-06-14:
### Describe the bug.

When I try to open the Sub GHZ Remote App it just shows "empty" and then when I try to leave it opens the data system folders. When I try to leave this it opens a UI where I can create a Map or something.

When I install the original Sub GHz Remote App it seems to work.

### Reproduction

1. Click the "Apps" Button.
2. Go into the "Sub GHz" Folder.
3. Open "Sub GHz Remote".

xMasterX, 2024-06-14:
What do you mean by "original" app?

and all what you described is not an issue, its empty because you have no map files in the folder, by clicking back it opens menu where you can make those map files or edit them
create map file, configure it, and app will not launch on <Empty> folder, because folder now will have your map files in it

closed 2024-06-14 by xMasterX

AztecCodes, 2024-06-14:
@xMasterX With the original App I mean the app you can install from the official App Store.

xMasterX, 2024-06-14:
> @xMasterX With the original App I mean the app you can install from the official App Store.

this is same app actually

AztecCodes, 2024-06-14:
@xMasterX Weird that it behaves differently then.