DarkFlippers/unleashed-firmware #764: BLE Spam doesn't work partially.

AztecCodes, 2024-06-14:
### Describe the bug.

Following BLE Spam features don't work when I tested it:
- BT Settings Flood
- iOS 17 Lockup Crash
- Android Device Connect
- Samsung Buds Popup

I couldn't test following features:
- Vibrate 'em all
- Denial of Pleasure

The other features work confirmed.
"The Kitchen Sink" works for the features that work.

### Reproduction

1. Click the "Apps" Button
2. Open the "Bluetooth" folder
3. Open the "BLE Spam" App
4. Try the different features out

xMasterX, 2024-06-14:
Hello, what do you mean "doesn't work" ?
If you are not aware, latest ios and android versions now includes fix for this
This is not a BUG

closed 2024-06-14 by xMasterX

AztecCodes, 2024-06-14:
@xMasterX Sorry for asking this here, however is there any place where I can ask something about the Discord Server? I can't find any other possibility to contact someone who is in charge.

xMasterX, 2024-06-14:
> @xMasterX Sorry for asking this here, however is there any place where I can ask something about the Discord Server? I can't find any other possibility to contact someone who is in charge.

What do you mean? You have been banned in our discord and can't join?

AztecCodes, 2024-06-14:
@xMasterX I am not sure. I can't see the Discord Server anymore in my list and the invitation doesn't work anymore. So I guess that I got banned.

xMasterX, 2024-06-14:
> @xMasterX I am not sure. I can't see the Discord Server anymore in my list and the invitation doesn't work anymore. So I guess that I got banned.

Share your discord username

AztecCodes, 2024-06-14:
@xMasterX azteccodes

xMasterX, 2024-06-14:
> @xMasterX azteccodes

you was banned for shit posting (spamming) in general channel

I'll give you 2nd chance, read server rules again after joining and follow them

Next time ban will be permanent

AztecCodes, 2024-06-14:
@xMasterX Thank you very much. This is very kind.