DarkFlippers/unleashed-firmware #768: JavaScript crashes when running `

eliandoran, 2024-06-23:
### Describe the bug.

The `widget.js` script from `js_examples` crashes immediately after executing it.

[image]

### Reproduction

1. Press OK to enter the menu.
2. Select Apps.
3. Select Scripts.
4. Select js_examples.
5. Press Up and select `widget.js`

### Target

unlshd-075

### Logs

```Text
259635 [I][Loader] Starting JS Runner
259638 [I][AnimationManager] Unload animation 'L1_Mods_128x64'
259656 [T][StorageApi] File/Dir 0000FDD8 alloc
259660 [T][StorageApi] File 0000FDD8 - 00010264 open (/ext/apps/Scripts/elian/widget.js)
259666 [T][StorageApi] File 0000FDD8 - 00010264 closed
259669 [T][StorageApi] File/Dir 0000FDD8 free
259675 [I][JS modules] Loading external module /ext/apps_data/js_app/plugins/js_widget.fal
259680 [T][StorageApi] File/Dir 00010050 alloc
259689 [T][StorageApi] File 00010050 - 000106D4 open (/ext/apps_data/js_app/plugins/js_widget.fal)
259763 [T][ApiHashtable] Can't find symbol with hash 7c96f087 @ 080AB32C!
259767 [T][ApiHashtable] Can't find symbol with hash 983a5ec1 @ 080AB32C!
259770 [T][ApiHashtable] Can't find symbol with hash 98383973 @ 080AB32C!
259774 [T][ApiHashtable] Can't find symbol with hash 4b64bd4d @ 080AB32C!
259778 [T][ApiHashtable] Can't find symbol with hash fdf5a8c7 @ 080AB32C!
259782 [T][ApiHashtable] Can't find symbol with hash ada6324e @ 080AB32C!
259785 [T][ApiHashtable] Can't find symbol with hash f65738c5 @ 080AB32C!
259789 [T][ApiHashtable] Can't find symbol with hash c9e52bce @ 080AB32C!
259793 [T][ApiHashtable] Can't find symbol with hash 4d4866a3 @ 080AB32C!
259797 [T][ApiHashtable] Can't find symbol with hash cba46b @ 080AB32C!
259800 [T][ApiHashtable] Can't find symbol with hash ed7500ce @ 080AB32C!
259804 [T][ApiHashtable] Can't find symbol with hash d85bd689 @ 080AB32C!
259808 [T][ApiHashtable] Can't find symbol with hash 549bd0a @ 080AB32C!
259811 [T][ApiHashtable] Can't find symbol with hash bc4d1c50 @ 080AB32C!
259815 [T][ApiHashtable] Can't find symbol with hash 962c4710 @ 080AB32C!
259819 [T][ApiHashtable] Can't find symbol with hash 922fe88 @ 080AB32C!
259823 [T][ApiHashtable] Can't find symbol with hash 2b03fe22 @ 080AB32C!
259826 [T][ApiHashtable] Can't find symbol with hash d91f014d @ 080AB32C!
259830 [T][ApiHashtable] Can't find symbol with hash d8b8ff20 @ 080AB32C!
259834 [T][ApiHashtable] Can't find symbol with hash d3ee8433 @ 080AB32C!
259837 [T][ApiHashtable] Can't find symbol with hash a9e82298 @ 080AB32C!
259841 [T][ApiHashtable] Can't find symbol with hash a02bb03f @ 080AB32C!
259845 [T][ApiHashtable] Can't find symbol with hash 83d1579a @ 080AB32C!
259849 [T][ApiHashtable] Can't find symbol with hash fdfcd38b @ 080AB32C!
259852 [T][ApiHashtable] Can't find symbol with hash dfd4830b @ 080AB32C!
259856 [T][ApiHashtable] Can't find symbol with hash 36912185 @ 080AB32C!
259860 [T][ApiHashtable] Can't find symbol with hash f11ed7d @ 080AB32C!
259864 [T][ApiHashtable] Can't find symbol with hash 3de00ec7 @ 080AB32C!
259867 [T][ApiHashtable] Can't find symbol with hash c07ab4dd @ 080AB32C!
259871 [T][ApiHashtable] Can't find symbol with hash d39ad3d @ 080AB32C!
259875 [T][ApiHashtable] Can't find symbol with hash 92fe76cf @ 080AB32C!
259879 [T][ApiHashtable] Can't find symbol with hash 54803eb8 @ 080AB32C!
259882 [T][ApiHashtable] Can't find symbol with hash d83f118e @ 080AB32C!
259886 [T][ApiHashtable] Can't find symbol with hash c9e513e8 @ 080AB32C!
259890 [T][ApiHashtable] Can't find symbol with hash af0c3fcc @ 080AB32C!
259893 [T][ApiHashtable] Can't find symbol with hash 30aa559e @ 080AB32C!
259897 [T][ApiHashtable] Can't find symbol with hash e7914ee4 @ 080AB32C!
259901 [T][ApiHashtable] Can't find symbol with hash e41634f2 @ 080AB32C!
259905 [T][ApiHashtable] Can't find symbol with hash 4ff5882 @ 080AB32C!
259908 [T][ApiHashtable] Can't find symbol with hash e3812d8b @ 080AB32C!
259912 [T][ApiHashtable] Can't find symbol with hash eb357866 @ 080AB32C!
259916 [T][ApiHashtable] Can't find symbol with hash 59c9920a @ 080AB32C!
259920 [T][ApiHashtable] Can't find symbol with hash 51e3ac48 @ 080AB32C!
259923 [T][ApiHashtable] Can't find symbol with hash 51e07f95 @ 080AB32C!
259927 [T][ApiHashtable] Can't find symbol with hash 8d97cd71 @ 080AB32C!
259931 [T][ApiHashtable] Can't find symbol with hash 8d8898b8 @ 080AB32C!
259934 [T][ApiHashtable] Can't find symbol with hash eb3537f4 @ 080AB32C!
259938 [T][ApiHashtable] Can't find symbol with hash 51dc1d30 @ 080AB32C!
259942 [T][ApiHashtable] Can't find symbol with hash 3702827f @ 080AB32C!
259946 [T][ApiHashtable] Can't find symbol with hash eb352f76 @ 080AB32C!
259949 [T][ApiHashtable] Can't find symbol with hash c9e55eda @ 080AB32C!
259953 [T][ApiHashtable] Can't find symbol with hash bc94c80a @ 080AB32C!
259957 [T][ApiHashtable] Can't find symbol with hash 221ca54 @ 080AB32C!
259961 [T][ApiHashtable] Can't find symbol with hash 42d0c164 @ 080AB32C!
259964 [T][ApiHashtable] Can't find symbol with hash d93acffc @ 080AB32C!
259968 [T][ApiHashtable] Can't find symbol with hash 6aac992f @ 080AB32C!
259972 [T][ApiHashtable] Can't find symbol with hash 5007b55a @ 080AB32C!
259976 [T][ApiHashtable] Can't find symbol with hash eb358b54 @ 080AB32C!
259979 [T][ApiHashtable] Can't find symbol with hash 5b9c541 @ 080AB32C!
259983 [T][ApiHashtable] Can't find symbol with hash 609d84a3 @ 080AB32C!
259987 [T][ApiHashtable] Can't find symbol with hash c4e5b9aa @ 080AB32C!
259990 [T][ApiHashtable] Can't find symbol with hash c4e0d2ba @ 080AB32C!
259994 [T][ApiHashtable] Can't find symbol with hash c4e72f74 @ 080AB32C!
259998 [T][ApiHashtable] Can't find symbol with hash 3af87667 @ 080AB32C!
260002 [T][ApiHashtable] Can't find symbol with hash bdd69f1b @ 080AB32C!
260006 [I][Elf] Total size of loaded sections: 6828
260009 [T][StorageApi] File 00010050 - 000106D4 closed
260011 [T][StorageApi] File/Dir 00010050 free
260014 [D][Fap] Library for js, API v. 1 loaded
260016 [D][Fap] Library for js, API v. 1 loaded
```

eliandoran, 2024-06-23:
Can reproduce the same issue with an even smaller script than `widget.js`:

```javascript
let widget = require("widget");

widget.addText(10, 10, "Primary", "Example JS widget");
widget.show();
```

All the other scripts seem to work fine. Did not test on the original firmware.

eliandoran, 2024-06-23:
@Willy-JL , I see you are quite involved in the JavaScript, maybe you might be willing to help me with this one.

Willy-JL, 2024-06-23:
I didn't make this module, Derek Jamison did in a PR to momentum fw. I have gotten one report of this in my discord community, given you found this too I'll move it up the priority list, thanks for the report!

Willy-JL, 2024-06-26:
@eliandoran i see the issue: view dispatcher was recently refactored in ofw to use event loop, this means some functions need to happen on same thread. precicely: enable_queue() and run() and free() must happen on same thread, which widget doesnt do

eliandoran, 2024-06-27:
@Willy-JL , thank you for the fix. I've tested it on the latest `dev` branch and it seems to be working fine.

PR #770 (2024-06-27): JS: Refactor widget and keyboard modules, fix crash
# What's new

- Fixes #768
- Uses more lightweight `ViewHolder` instead of `ViewDispatcher`
- Based on textbox and submenu refactors by nminaylov

# Verification 

- Example `widget.js` and `keyboard.js` doesn't crash, works as intended

# Checklist (For Reviewer)

- [x] PR has description of feature/bug
- [x] Description contains actions to verify feature/bugfix
- [ ] I've built this code, uploaded it to the device and verified feature/bugfix