DarkFlippers/unleashed-firmware #769: 075e - AirMouse does not respond to Keypad Input

Fboesch, 2024-06-24:
### Describe the bug.

On 074e - Airmouse does work correctly. On 075e - the Keypad Entry is unresponsive. Not even Exiting Application is possible: Only Flipper Reset will help.

### Reproduction

Start AirMouse (Adapter connected) - Use it: Mouse Cursor will move via Bluetooth. But no Keypad Entry (on Flipper) is responding. Not even "Back" Press for longer times.

xMasterX, 2024-06-27:
what exact app and module you are using?

can you provide photos or video of what happening?

Fboesch, 2024-06-27:
- Came from 074e and updated via qFlipper.  On 074e the App is named „[BMI160] Airmouse“ - on 075e Package the App shows a different name: „[BMI/LSM] Air Mouse“ in the GPIO Folder/Section. My Air Mouse Module is by Kasiin (http://kasiin.cn/archives/11.html)) - and works fine on 074e release. 
- on 074e the „[BMI160] Airmouse] Version works as intended (Keys (Mouse buttons) as well long press back to return to menu/exit app). 
- On 075e the (new?) App looks similar, but when entering in (Bluetooth Mouse Mode) - Mouse Cursor will work on attached PC/Tablet, but no Key Input (D-Pad or Back Button) works anymore. To exit the (running!) App Flipper needs to reset (Left + Back Button)….
- I just saw on my Flipper (Mobile) App that in the „App Store“ there seems to be an Update for the Air Mouse App. I installed it - but no change with 075e: No KeyPad/Back Button is reacting as soon the App is started.

Fboesch, 2024-06-27:
Reverting Flipper back to 074e - everything is working again (Also with updated BMI Air Mouse App from Flipper Mobile AppStore….
somehow the AirMouse App doesn‘t seem to be able to access the Key-Handler anymore on 075 Release…

xMasterX, 2024-06-27:
> Reverting Flipper back to 074e - everything is working again (Also with updated BMI Air Mouse App from Flipper Mobile AppStore…. somehow the AirMouse App doesn‘t seem to be able to access the Key-Handler anymore on 075 Release…

Also try on latest OFW dev with version from "appstore"

Fboesch, 2024-06-27:
I contacted  ginkage from Air Mouse and he confirms that my described issue is also on official FW:
https://github.com/ginkage/FlippAirMouse/issues/24

therefore: this issue is not specifically on Unleashed 🤗

closed 2024-06-27 by Fboesch