DarkFlippers/unleashed-firmware #778: Out of RAM error on NFC file launch from file browser

FalsePhilosopher, 2024-07-11:
### Describe the bug.

Using the file browser to run a NFC file will produce a out of RAM error.

### Reproduction

1. Open file browser
2. Scroll to NFC
3. Scroll to NFC file
4. Run the file
5. Out of RAM error

xMasterX, 2024-07-12:
Hi, steps to reproduce doesn't work, we are aware of this issue, its caused by RAM fragmentation which can be caused by anything, like any app, etc.

NFC app itself is too large and in such cases it can't fit when RAM is in this state

One user reported that they had NFC not starting even after reboot EVERY time, I suggested replacing/formatting their microsd and reinstalling firmware, and it somehow worked for them
If thats your case too, try this too

In my case I was able to reproduce this issue only 3 times and it was pretty random, currently we have no ideas how to make NFC app smaller to fit in all RAM conditions

AztecCodes, 2024-07-13:
@xMasterX 

In my case the Error occurs sometimes when I used the Flipper Zero before and used some apps and just do different things. The "Out of RAM Error" doesn't occur every time and also not very often, but it does happen from time to time. When I restart the Flipper and directly go for the NFC App, this error never occurs.
However when this Error doesn't occur, the NFC App takes an unusually long to load. It takes like 1-2 seconds till it opens up. It is not insanely long, but it is unusual and it wasn't like this in the older versions. All the other Apps open instantly, It takes always that long for the NFC App to load (If the Error doesn't occur). 

I couldn't find out how to reproduce it every time, but I think the chances that it happens are higher when you did use other apps for various things before opening the NFC App. 

My experience with your idea for a fix:
I did try out various things you recommended, however the issue still remained on the current release of Unleashed.
With older versions of Unleashed and the current official Flipper Software I don't experience this issue.

FalsePhilosopher, 2024-07-14:
I was getting furi crash errors on boot almost every boot from 73-75 and at random times before the out of ram errors. I formatted the sd card and reinstalled 75 from the SD update function. The furi crash errors at boot have gone away and I can launch a NFC file from the file browser, but like AztecCodes has stated that only work from a fresh boot. If you try to open a few ext apps then launch a nfc file from the file browser you will get a out of ram error. I have now gotten a busfault error instead of a out of ram error as well.

AztecCodes, 2024-07-14:
I can confirm that I get the busfault error too, but it only happened very few times.
It just happened right now, I didn't use the flipper for like 12 hours and it was on standby, now I tried to start the NFC App from the list and it directly rebooted and showed "busfault" as the error cause.

Danucosukosuko, 2024-08-21:
Yes, the same. I use apps before NFC, and the same error: Out of memory

Kf637, 2024-09-14:
I also have the same error.

```
255467 [I][Loader] Loading /ext/apps/NFC/nfc.fap
255625 [I][Elf] Total size of loaded sections: 76318
255628 [I][Loader] Loaded in 161ms
255642 [W][ViewPort] ViewPort lockup: see applications/services/gui/view_port.c:189
255677 [E][Elf] Not enough memory to load section data
255681 [E][Elf] Error loading section '.rodata'
255685 [E][Elf] Not enough memory
```

FalsePhilosopher, 2024-09-24:
Hold your back arrow for the quick power off function, does it display the crying dolphin or just turn off power? Yesterday I upgraded from 75e to 78e and the issue persisted. I noticed the different power off menu and went about fixing it. Rebooted into DFU mode, repaired to 1.0.1 from qflipper, once it booted after install I formatted the sd card. I then installed 78e from qflipper and I got my crying dolphin back an it doesn't crash launching files from explorer. I have not done extensive testing, just launched a file from browser and micro python which crashed with a bus error the first time but launched the second time.

Josue616, 2024-10-19:
That doesn't happen with xtreme firmware, but it was archived last July, so there will be no more updates for xtreme firmware, we will only talk about RAM error in NFC app, because xtreme firmware loads the .nfc after opening the list in app, not when you open the application directly.

AztecCodes, 2024-11-06:
@Josue616 You are right. It started to happen from one specific firmware version of Unleashed and Momentum on. I might be able to find out which one it was. Like the loading time was not as long it is now when starting the NFC app and it didn't crash.

Arsi94, 2025-01-18:
It seems I've found a solution to this issue. If there are too many NFC files in the folder where they are stored, the Flipper Zero goes into an error state. I moved some of the files I needed into a separate folder, and they launched without any problems. The solution is simply to distribute the NFC files across different folders in smaller amounts. I hope this helps someone!

Arsi94, 2025-01-18:
Here’s a PowerShell script to make this process faster. Thanks, DeepSeek.

# Укажите путь к корневой папке
$rootFolder = "C:\Users\"

# Получаем все файлы во всех вложенных папках
Get-ChildItem -Path $rootFolder -Recurse -File | ForEach-Object {
    # Пропускаем файлы, которые уже находятся в папках с их именами
    if ($_.Directory.Name -ne $_.BaseName) {
        # Создаем путь для новой папки (в той же директории, где находится файл)
        $newFolder = Join-Path -Path $_.DirectoryName -ChildPath $_.BaseName
        
        # Создаем папку, если она еще не существует
        if (-not (Test-Path -Path $newFolder)) {
            New-Item -Path $newFolder -ItemType Directory -Force
        }
        
        # Перемещаем файл в новую папку
        Move-Item -Path $_.FullName -Destination $newFolder
    }
}

AztecCodes, 2025-01-18:
@Arsi94 

Interesting! I did have many NFC files so this could help.
When I get my Flipper I will try it out.