DarkFlippers/unleashed-firmware #779: Error: Missing Imports

DefinetlyNotAI, 2024-07-12:
### Describe the bug.

Keep getting many fail applications with error missing imports and code err_04,

I am using the latest unleashed firmware UNLSHD-075

Many apps now fail to start, including ESP8266-Wifi-Deauther, ESP32-evil-portal and USB-HID-AutoFire,

### Reproduction

1. Take a flipper zero with the same applications as listed in the example
2. Have the latest firmware
3. Open the application, It will first say App Too Old APP:17 < FW:66
4. See the error 04

### Target

Applications (Any)

### Anything else?

Maybe not entirely related, but pressing left and clicking on the GPIO from the quick menu also says Update the framework, which is weird as the previous version i used had none of these issues: 073c

DefinetlyNotAI, 2024-07-12:
Woah, HUGE update, I found out this ONLY happens when you update the firmware!! Automatically many files become corrupt and from the structure of the sd card, the 2 firmware appear to be merged... Like both at the same time, when formatted then re-installed everything worked properly, What seems to also occur due to corruption is the SD cards memory CAP is set to the previous usage from the older version, so my SD CAP went from 64GB to measly 120MiB, I fixed it by reformating, dont think its easy to reproduce and might be a issue outside the firmware, but maybe just add like a wrapper check that logs the start cap of the SD card and compares it with the final reading, if it fails, reformat the SD card and output a error...

Thanks, appreciate the BLE spammer.

xMasterX, 2024-07-12:
Delete your `apps` folder and reinstall firmware, you have apps that you had added by yourself and not updated or your microsd is bad, also i recommend replacing your microsd with branded good one, like Kingston or SanDisk
you may loose all your files if some cheap microsd is being used in flipper

closed 2024-07-12 by xMasterX

DefinetlyNotAI, 2024-07-12:
My microsd card is kingston
It is a geniune, used a usb data test

xMasterX, 2024-07-12:
> My microsd card is kingston It is a geniune, used a usb data test

Try formatting it IN flipper settings and reinstalling firmware, nothing should be corrupt

DefinetlyNotAI, 2024-07-12:
Yeah thats what i did in the end

My issue was, is it from the firmware or the flasher?