DarkFlippers/unleashed-firmware #781: BadUSB keyboard layout feature doesn't work

AztecCodes, 2024-07-14:
### Describe the bug.

When I use BadUSB and select de-CH as the keyboard layout it somehow doesn't work. When I run the test payload for Windows which opens a text editor and writes this test message into it, the keyboard layout is clearly not de-CH. Also when I try other keyboard layouts they won't work neither. It is always the same layout. I guess the standard layout is the US layout?

The test message is mixed up with random symbols and it is clearly QWERTY and not QWERTZ (de-CH is QWERTZ).

I didn't have this problem on Release 73.

### Reproduction

1. Open BadUSB
2. Connect via USB to Windows 11 Computer
3. Select the "demo_windows" payload
4. Press left arrow to open configuration
5. Select the wanted keyboard layout
6. Run the payload

AztecCodes, 2024-08-03:
Little Note: It works fine on the current release of Momentum. mmtm-005

xMasterX, 2024-09-09:
Tested on mac with `fr-FR-mac` layout
All works fine
If wrong layout selected on flipper or pc - it makes "strange symbols" 
Feature works as it should, I can't reproduce a problem, maybe you used wrong layout?

[image]

closed 2024-09-09 by xMasterX

AztecCodes, 2024-09-16:
I did find out what the issue is.

This problem only appears when the text is printed in Notepad.
I tested it on multiple Windows 11 computers with the keyboard layout de-CH and it always was the same issue.
When I run the demo_windows on Notepad++ it worked fine.

Also all other payloads work fine (They weren't working because some related github file linked in the payload was deleted).

So it works perfectly fine, only in Notepad on Windows 11 this issue seems to appear.