DarkFlippers/unleashed-firmware #798: Remove non-functional pc_monitor app

TomXV, 2024-08-17:
## Issue
pc_monitor app fails to compile due to missing `furi_hal_bt.h` and is non-functional.

## Steps to Reproduce
1. Attempt to compile pc_monitor
2. Observe compilation error due to missing `furi_hal_bt.h`

## Impact
- App cannot be used
- May confuse users expecting functionality

## Proposed Solution
Remove pc_monitor from the package.

## Additional Notes
- Similar issue reported: https://github.com/TheSainEyereg/flipper-pc-monitor/issues/3
- Consider noting removal in changelog

Let me know if you need any more info or clarification.

xMasterX, 2024-08-17:
What source you are using to compile the app?
If you see already compiled app in unleashed firmware 'e' builds, why do you need to compile that yourself?
If you can open the app that included on the 'e' build, why do you say it cannot be used?

TomXV, 2024-08-17:
I used the app included in the Unleashed firmware 'e' build, but assuming that the version of the built application was old, I tried to clone the repository and compile the app's source code directly. I attempted to build using both the official SDK and this custom firmware, but the build fails. I believe this issue requires the app developer to redesign the code to be compatible with the latest SDK, but the app included in the Unleashed firmware 'e' build might be old and may not function correctly. Until then, I hope that either an alternative application is included in the package or the old version of pc_monitor is removed until it's updated to be compatible with the latest FW and SDK.

Reference: https://www.reddit.com/r/flipperzero/comments/16ltrpj/app_for_displaying_pc_resources/

xMasterX, 2024-08-17:
> I used the app included in the Unleashed firmware 'e' build, but assuming that the version of the built application was old, I tried to clone the repository and compile the app's source code [...]

Why do you think app is old?
It was fixed but not in the original repo by its original author
fixed version that is included in 'e' builds are based here
https://github.com/xMasterX/all-the-plugins/tree/dev/non_catalog_apps/pc_monitor

TomXV, 2024-08-17:
I see, it seems I misunderstood. I'm sorry.
To use this app, I compiled and ran the `flipper-pc-monitor-backend` mentioned in your README, but it doesn't work correctly.

Do you have any information about this?

xMasterX, 2024-08-17:
> I see, it seems I misunderstood. I'm sorry. To use this app, I compiled and ran the `flipper-pc-monitor-backend` mentioned in your README, but it doesn't work correctly.
> 
> Do you have any information about this?

I'm not sure whats the problem with backend you have
How exactly it doesnt work correctly?

TomXV, 2024-08-17:
When I run the flipper-pc-monitor-backend, I get the following error:

```
thread 'tokio-runtime-worker' panicked at src\helpers.rs:17:10:
called `Result::unwrap()` on an `Err` value: ParseIntError { kind: InvalidDigit }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

xMasterX, 2024-08-17:
> When I run the flipper-pc-monitor-backend, I get the following error:
>
> ```
> thread 'tokio-runtime-worker' panicked at src\helpers.rs:17:10:
> called `Result::unwrap()` on an `Err` value: ParseIntError [...]

I can't help with some problem in Rust 3rd party lib in 3rd party app

But you can try compiling this version from PR
https://github.com/eliandoran/flipper-pc-monitor-backend/tree/dev/cargo_lock
It has cargo lock so packages will use correct versions
Also you can try using pre compiled builds from the release

TomXV, 2024-08-17:
Thank you for your assistance. I tried building the repository you provided and running the pre-built files, but I kept encountering the same error:

```
thread 'tokio-runtime-worker' panicked at src\helpers.rs:17:10:
called `Result::unwrap()` on an `Err` value: ParseIntError { kind: InvalidDigit }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

After further investigation, I found an explanation for this issue in a comment on another thread:

https://github.com/TheSainEyereg/flipper-pc-monitor-backend/issues/10#issuecomment-1929651303

This explanation clarifies that the issue is related to GPU drivers, particularly for Nvidia and AMD GPUs. 

Given this understanding, I believe we can now close this issue. The problem appears to be a known limitation with certain GPU configurations, and the solution involves updating GPU drivers or addressing potential compatibility issues with AMD GPUs.

Thank you for your time and assistance in addressing this matter.

closed 2024-08-17 by TomXV