DarkFlippers/unleashed-firmware #806: Update Failed: Manifest pointer error

3DPixel, 2024-09-11:
### Describe the bug.

During update of release 0.78, I get an error message during update:

Update Failed!
Manifest pointer error
#[1-20]

Retry or abort

### Reproduction

1. Went to the releases page https://github.com/DarkFlippers/unleashed-firmware/releases/tag/unlshd-078
2. Clicked on 'Default' under 'Web Updater'
3. Update started, browser handed off update to the Flipper.
4. Update subsequently failed.

After aborting, another error message 'No database found on SD card'. Tried reflashing firmware. same problem.
Occasionally, I get 'SD card not found' error also after failed update.

3DPixel, 2024-09-11:
Flipper Model: FZ.1

If the firmware is flashed via the desktop app, it is successful.

xMasterX, 2024-11-11:
Not related to the firmware, issue is caused by other cases, faulty sd card, bad formatting on sd card, companion app isssue, etc.

closed 2024-11-11 by xMasterX