DarkFlippers/unleashed-firmware #809: No BT connection from Iphone on OF 1.0.1 + UL 0.78

Dmitry422, 2024-09-18:
### Describe the bug.

When I use OF 1.0.1 - BT connection to flipper work fine.
After update to UL 0.78 - iPhone and PC can see flipper by bluetooth.
Flippers BT module working - I test it with BLE software.

Not sure, but I must attach debug log for this issue ?

### Reproduction

Upgrade to UL 0.78 from OF 1.0.1

### Target

BT, bluetooth, iPhone, qflipper

xMasterX, 2024-09-18:
https://github.com/DarkFlippers/unleashed-firmware/blob/dev/documentation/FAQ.md#i-installed-unleashed-firmware-and-now-my-mobile-app-doesnt-connect-to-flipper--or-i-changed-flipper-device-name-and-my-mobile-app-now-doesnt-connect-to-flipper-

closed 2024-09-18 by xMasterX

Dmitry422, 2024-09-18:
Hello, xMasterX!
I know this solution, but it’s not working for me.
I’m unpair all devices from F0 and remove F0 from my phone BT devices. But still F0 not available by BT not on my phone, not on another iPhone or Mac pc. This is the reason why I create bug report and ask about debug.

Best Regards, 
Thank you!

xMasterX, 2024-09-18:
> Hello, xMasterX! I know this solution, but it’s not working for me. I’m unpair all devices from F0 and remove F0 from my phone BT devices. But still F0 not available by BT not on my phone, not on [...]

It should be possible to connect only via flipper official app, not via regular bluetooth menu on ios
Except if you are using app for remote control (keyboard, mouse) on flipper, it makes new connection with different parameters and ios shows that in regular menu

Dmitry422, 2024-09-18:
Thank you!
This way working !!