DarkFlippers/unleashed-firmware #817: Sub GHz App freezes when I use Frequency Analyzer and select the found signal when Modulation FM238 is set

AztecCodes, 2024-10-07:
### Describe the bug.

**Device Information:**
Firmware: unlshd-078
Model: FZ.1
SD Card Format: exFAT

**Sub-GHz Settings:**
Module: Internal (Bug doesn't occur with external antenna)
Modulation: FM238
Bin RAW: ON (I don't think this matters)

**The Bug:**
When I open the Sub GHz Application and start the Frequency Analyzer I send a signal and the Frequency Analyzer shows the Frequency. I click on OK Button then it reads in this frequency. At this point the Flipper freezes. It doesn't even crash, it just freezes. I have to force reboot it by using button combination. This only happens with Modulation FM238 and only with internal antenna in my case.

### Reproduction

1. Reboot Flipper Zero
2. Select Sub GHz Application in Menu (Or start it from File System)
3. Select "Read" and go to config and set Modulation to FM238. Also turn Bin RAW to ON.
4. Now use the return button to go back to the menu of the Sub GHz Application
5. Select "Frequency Analyzer"
6. Send a signal or wait till your receive a random signal so the Frequency Analyzer shows a Frequency
7. Click the OK Button so it starts the read function with the found frequency
8. At this point the Flipper should freeze. You can't do anything and you have to force reboot with button combination.

### Anything else?

I tested every Modulation and it only happens with FM238.
When I use this external antenna: [Flipper Zero External CC1101 Module](https://www.tindie.com/products/flipmodules/flipper-zero-external-cc1101-module/)
the bug doesn't appear.

If more information is necessary feel free to ask me!

xMasterX, 2024-10-07:
Hello, try latest dev build, issue is not related to modulation, it was due to incorrect scene logic, I fixed that in dev branch and it will be in next release

AztecCodes, 2024-11-06:
@xMasterX The bug still occurs on Unleashed dev build 1069 and also on Momentum dev build 3EF28382.

closed 2024-11-11 by xMasterX