DarkFlippers/unleashed-firmware #818: NFC Magic crash crash/reboot: furi_check failed

AlfaJackal, 2024-10-09:
### Describe the bug.

Custom unlshd-078e

Writing and wiping results in a reboot/crash.

### Reproduction

1. Open NFC Magin
2. Check Magic Tag
3. Choose More
4. Choose Write or Wipe
5. Choose saved nfc file

### Logs

```Text
Flipper crashed and was rebooted

furi_check failed
```

webmentors, 2024-10-18:
Same on momentum mtm-007

diggidong, 2025-01-06:
Same on `unlshd-079e`.

Temporary fix is to change the name manually.
Only preset file name crashes the save process.

ilpanich, 2025-01-10:
> Same on `unlshd-079e`.
> 
> Temporary fix is to change the name manually. Only preset file name crashes the save process.

Same for me...

ar2g20, 2025-04-09:
Can confirm this still happens on the latest firmware, with a card dump named "Full"

lelvisl, 2025-08-03:
same 082e

Markb1337, 2025-08-20:
Oh and for what it's worth, I was able to write the card succesfully using the 'MIFARE Classic Tool' Android app (https://play.google.com/store/apps/details?id=de.syss.MifareClassicTool)

Markb1337, 2025-08-21:
Just hooked up the Flipper CLI with logging set to 'Trace' on the Flipper. The log output is below.

What's weird is that when the Flipper is connected to the Flipper CLI, the Flipper does not crash and reboot to the 'furi_check failed' screen. Instead it just hangs on the writing screen and gets completely stuck right there.

```
Firmware version: release-cfw unlshd-082e (5d085442 built on 16-07-2025)

>: log
Current log level: trace
Use <log ?> to list available log levels
Press CTRL+C to stop...
37729 [D][BrowserWorker] End
37744 [D][DolphinState] icounter 411, butthurt 0
37749 [I][AnimationManager] Unload animation 'L1_Doom_128x64'
37761 [T][StorageApi] File/Dir 00002C88 alloc
37763 [I][Loader] Loading /ext/apps/NFC/nfc_magic.fap
37767 [T][StorageApi] File 00002C88 - 0000A014 open (/ext/apps/NFC/nfc_magic.fap)
37861 [I][Elf] Total size of loaded sections: 34040
37864 [T][StorageApi] File 00002C88 - 0000A014 closed
37866 [T][StorageApi] File/Dir 00002C88 free
37869 [I][Loader] Loaded in 106ms
44277 [D][ViewDispatcher] View changed while key press 2000EDE8 -> 2000F058. Sending key: OK, type: Release, sequence: 000000EC to previous view
44295 [D][Nfc] FWT Timeout
44297 [T][Nfc] Failed TRX state machine
44330 [D][Nfc] FWT Timeout
44331 [T][Nfc] Failed TRX state machine
44456 [D][Nfc] FWT Timeout
44458 [T][Nfc] Failed TRX state machine
44586 [D][Nfc] FWT Timeout
44587 [T][Nfc] Failed TRX state machine
44714 [D][Nfc] FWT Timeout
44716 [T][Nfc] Failed TRX state machine
44739 [D][Nfc] FWT Timeout
44740 [T][Nfc] Failed TRX state machine
44792 [D][Nfc] FWT Timeout
44794 [T][Nfc] Failed TRX state machine
44796 [T][FelicaPoller] Activation failed error: 8
44822 [D][Nfc] FWT Timeout
44824 [T][Nfc] Failed TRX state machine
44847 [D][Nfc] FWT Timeout
44848 [T][Nfc] Failed TRX state machine
44972 [D][Nfc] FWT Timeout
44974 [T][Nfc] Failed TRX state machine
45097 [D][Nfc] FWT Timeout
45098 [T][Nfc] Failed TRX state machine
45224 [D][Nfc] FWT Timeout
45225 [T][Nfc] Failed TRX state machine
45254 [D][Nfc] FWT Timeout
45255 [T][Nfc] Failed TRX state machine
45307 [D][Nfc] FWT Timeout
45309 [T][Nfc] Failed TRX state machine
45311 [T][FelicaPoller] Activation failed error: 8
45341 [D][Nfc] FWT Timeout
45343 [T][Nfc] Failed TRX state machine
45369 [T][ISO14443_3A] Sel resp: 08
45370 [T][ISO14443_3A] Col resolution complete
45374 [D][GEN2] ATS request succeeded:
45398 [T][ISO14443_3A] Sel resp: 08
45400 [T][ISO14443_3A] Col resolution complete
45418 [D][Nfc] FWT Timeout
45419 [T][Nfc] Failed TRX state machine
45444 [T][ISO14443_3A] Sel resp: 08
45446 [T][ISO14443_3A] Col resolution complete
55531 [D][ViewDispatcher] View changed while key press 20013270 -> 2000EDE8. Sending key: Right, type: Release, sequence: 000000F0 to previous view
56005 [D][BtSrv] call 1, lock 0x00000000, result 0x00000000
56011 [D][GattChar] Updating Battery Level char
58798 [D][BrowserWorker] Start
58801 [W][Storage] /any is deprecated, use /ext instead
58808 [W][Storage] /any is deprecated, use /ext instead
58811 [T][StorageApi] File/Dir 00014C50 alloc
58813 [W][Storage] /any is deprecated, use /ext instead
58817 [T][StorageApi] Dir 00014C50 - 00014D2C open (/any/nfc)
58827 [T][StorageApi] Dir 00014C50 - 00014D2C closed
58830 [T][StorageApi] File/Dir 00014C50 free
58832 [D][BrowserWorker] Enter folder: /any/nfc items: 15 idx: -1
58836 [D][BrowserWorker] Load offset: 0 cnt: 50
58840 [T][StorageApi] File/Dir 00014C50 alloc
58845 [W][Storage] /any is deprecated, use /ext instead
58848 [T][StorageApi] Dir 00014C50 - 00014D2C open (/any/nfc)
58867 [T][StorageApi] Dir 00014C50 - 00014D2C closed
58871 [T][StorageApi] File/Dir 00014C50 free
66007 [D][BrowserWorker] End
66016 [W][Storage] /any is deprecated, use /ext instead
66020 [T][StorageApi] File/Dir 00012F88 alloc
66024 [W][Storage] /any is deprecated, use /ext instead
66030 [T][StorageApi] File 00012F88 - 000143A4 open (/any/nfc/Afvalpas.nfc)
66086 [T][StorageApi] File 00012F88 - 000143A4 closed
66089 [T][StorageApi] File/Dir 00012F88 free
66092 [W][Storage] /any is deprecated, use /ext instead
66098 [T][StorageApi] File/Dir 00012F50 alloc
66101 [W][Storage] /any is deprecated, use /ext instead
66105 [T][StorageApi] File 00012F50 - 0001432C open (/any/nfc/assets/mf_classic_dict_user.nfc)
66120 [T][KeysDict] Read line: FA6D910CA4A5
                                           , len: 13
66123 [I][KeysDict] Loaded dictionary with 1 keys
66134 [D][ViewDispatcher] View changed while key press 2000EDE8 -> 20013530. Sending key: OK, type: Release, sequence: 000000F4 to previous view
66143 [T][ISO14443_3A] Sel resp: 08
66145 [T][ISO14443_3A] Col resolution complete
66258 [T][ISO14443_3A] Sel resp: 08
66260 [T][ISO14443_3A] Col resolution complete
66264 [D][MfClassicPoller] 1K detected
```

At this point the Flipper freezes and I have to press a key combo to get it to reboot.

closed 2026-07-01 by mishamyte