DarkFlippers/unleashed-firmware #820: Frequency Analyzer crashes: "furi_check failed"

min2sia, 2024-10-11:
### Describe the bug.

Frequency Analyzer crashes at random time intervals. Sometimes it can run several hours without crashing, sometimes just a few minutes.

FZ.1
unlshd-078e (...) release-cfw

### Reproduction

Nothing fancy, just go to Sub-GHz -> Frequency Analyzer.

min2sia, 2024-10-12:
Tried capturing logs via https://lab.flipper.net/. When it crashes all I can see is:
```
...
16:0:55 [DEBUG] Device: guiStartScreenStream: OK
16:0:55 [DEBUG] Main: systemSetDatetime: OK
21:10:45 [INFO] Main: Flipper has been disconnected
21:10:49 [INFO] Main: Flipper connected
21:11:7 [ERROR] Main: Error while starting RPC: RPC timeout: systemPingRequest
21:11:7 [INFO] Main: RPC started
21:11:23 [ERROR] Device: RPC error in command 'guiStopScreenStream': RPC timeout: guiStopScreenStreamRequest
21:11:23 [DEBUG] Device: guiStopScreenStream: OK
```
I have set log level to "Trace", Log device - USART, Debug - ON, Heap trace - Main.

Any hints how to troubleshoot this crash?

min2sia, 2024-10-15:
Update: reverted to stock firmware and Frequency Analyzer works fine there. Running for couple of days with no crashes.

xMasterX, 2024-10-15:
Please stop your investigation and read my reply here in first issue, this one will be closed as duplicate later:
https://github.com/DarkFlippers/unleashed-firmware/issues/817#issuecomment-2396790185

closed 2024-11-11 by xMasterX