DarkFlippers/unleashed-firmware #831: Hi ,how i cand self buil firmware with all extra?

girs1982, 2024-11-06:
### Description of the feature you're suggesting.

how this
[image]

xMasterX, 2024-11-07:
https://github.com/DarkFlippers/unleashed-firmware/blob/462ab255a16a6edf9473ddad0990f5c52baa7861/.drone.yml#L65

girs1982, 2024-11-07:
> Follow link below and see how build is made:
> 
> https://github.com/DarkFlippers/unleashed-firmware/blob/462ab255a16a6edf9473ddad0990f5c52baa7861/.drone.yml#L65

THNX, but what need modificate?

Dmitry422, 2024-11-26:
> > Follow link below and see how build is made:
> > https://github.com/DarkFlippers/unleashed-firmware/blob/462ab255a16a6edf9473ddad0990f5c52baa7861/.drone.yml#L65
> 
> THNX, but what need modificate? and how it run? ./fbt & what? or need docker? and another question ,how i can crypt my user manafacture? how it done in yours builds

Hi! In this file you can see console command sequence and can see what happen with source code before building.
Check under section "Build with extra apps".
In short - you must download extra apps pack, unzip to firmware folder and build with ./fbt

girs1982, 2024-11-26:
> Hi! In this file you can see console command sequence and can see what happen with source code before building. Check under section "Build with extra apps". In short - you must download extra apps [...]

Need more details

Dmitry422, 2024-11-26:
Sorry, mate! 
If you can't read simple instructions with console commands from file - so may be you not ready to compile your own firmware. 

Ask what exactly you not understand.

girs1982, 2024-11-26:
> Sorry, mate! If you can't read simple instructions with console commands from file - so may be you not ready to compile your own firmware.
> 
> Ask what exactly you can't understand.

need folders , and command ,all easy , no sorry

Dmitry422, 2024-11-26:
drone.yml file - its config file for DRONE service. So if you have not DRONE installed then you must do it manually - by command line. If you have DRONE installed then you can do it automatically.

So just repeat this command in console and enjoy

Dmitry422, 2024-11-26:
wget https://github.com/xMasterX/all-the-plugins/releases/latest/download/all-the-apps-extra.tgz
tar zxf all-the-apps-extra.tgz
mkdir -p applications/main/clock_app/resources/apps
cp -R extra_pack_build/artifacts-extra/* applications/main/clock_app/resources/apps/
rm -rf extra_pack_build
rm -f build/f7-firmware-C/toolbox/version.*
./fbt COMPACT=1 DEBUG=0 updater_package
mkdir artifacts-extra-apps
mv dist/f7-C/* artifacts-extra-apps/

girs1982, 2024-11-26:
> drone.yml file - its config file for DRONE service. So if you have not DRONE installed then you must do it manually - by command line. If you have DRONE installed then you can do it automatically.
> 
> So just repeat this command in console and enjoy

what the command ? concretly

Dmitry422, 2024-11-26:
> wget https://github.com/xMasterX/all-the-plugins/releases/latest/download/all-the-apps-extra.tgz tar zxf all-the-apps-extra.tgz mkdir -p applications/main/clock_app/resources/apps cp -R [...]

this commands from - I take it from file drone.yml

Mate ))) you not ready for build you own firmware ) just take compiled release or dev from git and relax.

Dmitry422, 2024-11-26:
I hope you not on windows OS ? Here commands for *nix systems.

girs1982, 2024-11-26:
> I hope you not on windows OS ? Here commands for *nix systems.

it not your problem ,ready or not, have ./fbt comman for it?

girs1982, 2024-11-26:
> > > I hope you not on windows OS ? Here commands for *nix systems.
> > 
> > 
> > it not your problem ,ready or not, have ./fbt comman for it?
> 
> yes its not my problem ) have a nice time!

maybe you rooster sima? , it was simple question

girs1982, 2024-11-26:
and why i need lose my time ,to  see sorce code ,for simple compilations?

Dmitry422, 2024-11-26:
> and why i need lose my time ,to see sorce code ,for simple compilations?

its because you are rooster sima ))))))

girs1982, 2024-11-26:
> > and why i need lose my time ,to see sorce code ,for simple compilations?
> 
> its because you are rooster sima ))))))

you rooster stupid, i asked only one qestion , what the command do it!?, and you stupid rooster ,do flood , and do shit without your mouth

girs1982, 2024-11-26:
and your cockerel opinion, who is ready for what, is not at all interesting to anyone, you were asked only one team, a stupid cockerel, are you afraid of losing your bread on these firmware!?

Dmitry422, 2024-11-26:
its your resume, stupid rooster:
```
Возможна постройка серверов на ваших системных блоках,таких как Apache2,Apache-Tom-Cat,
Python,Python flask,ANTI DDOS защита,SSL защита...и многое другое
ФРЕЙМ ВОРКИ И ЯЗЫКИ:
C,C++,C#,PHP,PYTHON,NODE-JS,GO,ANDROID-JAVA,java,javaserial,javascript,delphi,Gstreamer,QT,Angular,jQuery,Ajax
МИКРОКОНТРОЛЛЕРЫ :
esp8266,esp32,stm32,pic,avr,arduino,
delta electronics,raspberry pi,LG PLC,LG K120S (K7M-DR20UE),
и все остальные...

```
just go and fuck yourself, stupid cocksucker ))))

Dmitry422, 2024-11-26:
you are not only stupid - you are blind stupid rooster ))) I was your spoiler ))))
Cringe ))))))