DarkFlippers/unleashed-firmware #848: [Sub-Ghz] Support for Keeloq HCS362 and all parameters

yashikada, 2024-12-22:
### Description of the feature you're suggesting.

I have a keyfob with HCS362 and the following configuration:

- TE 200uS
- modulation Manchester
- trasmission 8 words
- 868.3M (EU)

[Here](https://github.com/merbanan/rtl_433/issues/3112) you can found more captures with a RTL-SDR.
This keyfob is not supported by FZ, I can see only BinRAW.
I have a pickit3 and I tried to change the content of EEPROM (the keyfob is dismissed, I can lost the keys) and all changes are not supported by FZ. 
TE: 100, 200, 400, 800 uS or Modulation: PWM, Manchester.

I'd like to see it supported and I'd like to see support for generic Keeloq, HCSxxx devices are configurable and generic support should be good than specific configuration for each implementation.
Keeloq protocol is one with many configurations.

Crsarmv7l, 2025-01-09:
Keeloq is a generic implementation. All keeloq have the same baud and preamble. The encrypted portion is put through all of the MFR codes until one is deemed valid.

Without a MFR code there is literally nothing to be done and in order to get one you need to pull it from the remote and submit it.

yashikada, 2025-01-09:
> All keeloq have the same baud

the baud rate of the HCS362 is configurable between 100 and 800 uS, the keyfob I had has the baud rate configured at 200 uS while the FP only detects 400 uS.

Crsarmv7l, 2025-01-10:
Oh very interesting! Thanks for explaining, as I haven't played with one of
those.

Learned something new today