DarkFlippers/unleashed-firmware #849: [Question] Does it support MIFARE Classic 1KB 13.25MHZ?

Drjacky, 2024-12-23:
### Description of the feature you're suggesting.

Support MIFARE Classic 1KB 13.25MHZ

Grim1313, 2025-02-05:
Of course it does.
I suspect that you are unlikely to find a firmware that does NOT support it right now :)

[image]

Drjacky, 2025-03-14:
Nice!

Read and write? @Grim1313 
I want to overwrite Fob B over Fob A.

Grim1313, 2025-03-14:
> Read and write?
For sure, read and write.
> I want to overwrite Fob B over Fob A.
Of course, the target mifare tag must support "magic" functions, that is, to put it simply, it must be rewritable.

Drjacky, 2025-03-22:
Could you please help me to find the right Blank/Rewritable Fob on Amazon?
I [found](https://www.amazon.co.uk/HFeng-13-56MHz-Changeable-Rewritable-Writable/dp/B07DX2Y2KV?th=1) a few that only Block 0 is rewritable. I'm quoting from someone who was working on a software to write on Fobs:
> These fobs are as described. Block zero (0) is 100% rewriteable, that does include the UID. I was trying to rewrite these with some software I wrote myself wondering why it wasn't working at first but I also learned that these Chinese-made fobs need a certain "unofficial" sequence of bytes sent to it in order to unlock block zero which are in fact used in certain libraries. Once I figured that needed to be done I was able to write whatever data to it I wanted.

Grim1313, 2025-03-22:
I think [this product](https://www.amazon.co.uk/Writable-Rewritable-Compatible-Classic-Control/dp/B0DHZRKC2K/) is more suitable. However, to be sure that modern readers will accept the fob, try to find an item where the description or reviews say that it is "Gen2", because "Gen1" is outdated and does not guarantee 100% opening.

Drjacky, 2025-03-23:
> [@Drjacky](https://github.com/Drjacky) I think [this product](https://www.amazon.co.uk/Writable-Rewritable-Compatible-Classic-Control/dp/B0DHZRKC2K/) is more suitable. However, to be sure that modern readers will accept the fob, try to find an item where the description or reviews say that it is "Gen2", because "Gen1" is outdated and does not guarantee 100% opening.

Thanks!
this one says
> These aren't anything special, just standard "Gen1a" UID-changeable keyfob tags. They're compatible with the Proxmark3, Flipper, and a variety of specialized RFID cloners, but not compatible with the popular Android MCT app.

Grim1313, 2025-03-23:
@Drjacky 
Well, it should work anyway.  Gen2 fobs are more expensive and harder to find on sale on marketplaces.

Drjacky, 2025-04-12:
I bought this one https://www.amazon.ie/product-reviews/B0BQ38R3TX
Someone posted a video as a review there, using Flipper Zero:

[image]

I have the official FW yet, but it seems they have added the capability to read/write Mifare classic: https://github.com/flipperdevices/flipperzero-firmware/releases
, and literally it worked. I used, Read and saved the data; and then used [NFC Magic](https://lab.flipper.net/apps/nfc_magic), and wrote the data into that Fob.

Then to make sure it wrote it successfully, I Read it again using original NFC app, and it displayed all the data as I read it in first place.

But when I tapped the Fob to the reader, reader didn't even recognize it; meaning not even beep an error. But I tapped the fob to another type of reader (for postbox which as totally a different fox - not Mifare Classic) and that reader recognized the fob and beep an error. I mean there is some data into it (not only because second reader recognize it but also, I read it using FZ and it displayed all the original fob data).
So, my question is,
1- Do I still need to use this Unleashed FW? or do you agree the official FW supports MiFare Read/Write/Crack?
2- Do you have any thoughts why the reader  doesn't even recognize it?

P.S. By original NFC app, I mean:
[image]

P.S. I didn't erase the target Fob first before writing into it. Had I?

Thank you in advance @Grim1313 🙏🏼

Drjacky, 2025-04-12:
Okay, seems I missed this step:
- https://flipper.wiki/mifareclassic/#reading-the-card
as the Read didn't receive 32/32 16/16:
[image]

I'll comment after [MFKey](https://docs.flipper.net/nfc/mfkey32) is done...

Drjacky, 2025-04-13:
[With access to the reader and card](https://docs.flipper.net/nfc/mfkey32#tdlq0) it worked! 🎉

closed 2025-04-13 by Drjacky