DarkFlippers/unleashed-firmware #850: Auto lock disable

Im-CatDev, 2024-12-23:
### Description of the feature you're suggesting.

Hi,

I would like to suggest adding an option that will automatically disable auto locking on the flipper when it's connected to qFlipper or Flipper Lab. It is very annoying when you are doing something through these apps and the device suddenly gets locked and disconnects. At the moment you have to constantly click buttons on the Flipper so it doesn't lock or disable the autolock manually every time you connect to the computer, and reenable it when you disconnect it. These options are very inconvienient so i think that it would be very useful

### Anything else?

It could be toggleable in settings

Dmitry422, 2024-12-31:
In progress.

Dmitry422, 2025-01-15:
I just wanna say: We add option to desktop settings "USB disable autolock" - now its not fully complete (work only ON/OFF), but in most cases it can help you disarm auto lock when F0 connect to PC (charged). I will continue this task and try add checking "is F0 connected to some APPs" - it will be option named like "RPC".

Dmitry422, 2025-01-17:
Additional info. We make some changes so now device will not be auto_locked if it have **active** USB connection **instead of** UBS **charging** connection. For BT connection (check by RPC session) still in project.

I hope this changes will be in 1082 dev build.

Im-CatDev, 2025-01-19:
What do you think about commuting that to ofw?

Dmitry422, 2025-01-19:
I write it based on Unleashed firmware + I have small experience with GitHub commits and etc (im newby here) - so I think its will be complicated for me now.

 so I think if MMX (unleashed owner) can make commit to OFW with my commit - I won't mind.

Im-CatDev, 2025-01-19:
@xMasterX can you look at this?

xMasterX, 2025-01-19:
> [@xMasterX](https://github.com/xMasterX) can you look at this?

Yes, a bit later

Im-CatDev, 2025-01-19:
> [@Im-CatDev](https://github.com/Im-CatDev) Why its important for you ? Just use Unleashed and enjoy )))

It would help more people, wouldn't it?

Dmitry422, 2025-01-19:
I don't mind ) so why not ? )