DarkFlippers/unleashed-firmware #852: [Sub-GHz] KeeLoq-Gorke standard support

mattcabb, 2025-01-01:
### Description of the feature you're suggesting.

Wanted to add flipper to my existing remote gate setup but it looks like Gorke Electronics (PL) is using some kind of custom modification of the KeeLoq standard. Would love to see it implemented into flipper.

### Anything else?

From scanning original remote I can see that the signal is 433MHz 64-bit. Not sure what kind of data is needed to develop support for Gorke devices.

Nigdzie, 2025-01-04:
Maybe a photo of remote, inside outside and product page would help a bit.

mattcabb, 2025-01-07:
Some photos of the remote + 10 `.sub` files from READ (10x single press of button #4 ).
Is there anything else that I can provide?

[image]
[image]
[image]
[image]
[image]
[image]
[image]

[gorke_singleButton_10reads.zip](https://github.com/user-attachments/files/18332063/gorke_singleButton_10reads.zip)

mattcabb, 2026-09-07:
@xMasterX  Hi! Attaching requested RAW captures.

[Gorke-RAWs.zip](https://github.com/user-attachments/files/31917881/Gorke-RAWs.zip)

xMasterX, 2026-09-07:
Tested, bad news, still no mfkey for this system, sadly, maybe it will appear in the future!

mattcabb, 2026-09-07:
Thanks for testing it. Hope it will mov forward one day.