DarkFlippers/unleashed-firmware #854: The back button hangs when using a subghz file

Crsarmv7l, 2025-01-08:
### Describe the bug.

Pressing the back button is not always responsive when using a Subghz file. This is not the case in any other app. It is worst with RAW's but still exists when using decoded protocols.

Subghz menus are fine. It only appears after selecting a file to playback.

### Reproduction

Favorites or subghz via a shortcut, select a file, press back.

Dmitry422, 2025-01-15:
Not sure, but it can be not bug but a future. Check with long press back button.

Crsarmv7l, 2025-01-15:
> Not sure, but it can be not bug but a feature. Check with long press back
> button.

Definitely not a feature when it takes multiple presses, long presses, and
it still may not exit.

Dmitry422, 2025-01-15:
I try replay your problem in dev.build - but I can't.
From shortcuts and from sub-ghz app  - short press not exit (its normal) and long press exit every time when I long press key.

Crsarmv7l, 2025-01-15:
https://imgur.com/a/yLKEzgx

Video of the issue

Dmitry422, 2025-01-15:
I sow 2-3 short (not exit) and after 1 long press (exit). Its right ?
Looks like all work correct.

Crsarmv7l, 2025-01-15:
Long doesn't always exit. Seriously, with the Tesla raw, when the bug
happens, it makes me wonder if I will have to let the battery die.

In addition why would MMX deviate from short press exit that is in every
single other app, and if he did why is it not in any changelog?

I don't understand why you are trying so hard to make aberrant behavior
seem like it's "correct".

I have been on unleashed since v 1x. This behavior is not "normal"

Crsarmv7l, 2025-01-15:
Here you go.

Multiple long presses doing nothing. This is a bug.

https://imgur.com/a/6EDsdLO

Dmitry422, 2025-01-15:
1. Did you try dev builds ? Lets try exclude bug in release - try last dev build - 1081E
2. Can you attach here you file - I will test it on my F0
3. I don't know why exactly long press for exit was used, but I think for this exist some reasons.
4. I don't try catch you, joke with you or something else - I just say that in some cases long press exit button its normal and your first video looks normal.

Finally we can make a tests - you try dev build and me try you file. And after we can find right answer for this bug.

Crsarmv7l, 2025-01-15:
> 1. Did you try dev builds ? Lets try exclude bug in release - try last dev build - 1081E
> 2. Can you attach here you file - I will test it on my F0
> 3. I don't know why exactly long press for exit was used, but I think for this exist some reasons.
> 4. I don't try catch you, joke with you or something else - I just say that in some cases (some apps) long press exit button its normal and your first video looks normal.
> 
> Finally we can make a tests - you try dev build and me try you file. And after we can find right answer for this bug.

1. I have not, but I will. Might take a few days
2. File attached. Rename txt to .sub. As I said multiple times; it does not happen every time. But it does happen pretty frequently.
3.  I highly doubt MMX changed the behavior of the button between 73 and current. But anything is possible. When the button is recognized a short press is enough to go back.

[Tesla_Charge_Port_315MHz_AM270.txt](https://github.com/user-attachments/files/18429389/Tesla_Charge_Port_315MHz_AM270.txt)

Dmitry422, 2025-01-17:
Hello!
I test your file about 3-4 times with 10-15 repeats every tests - no one problem, exit every time on short press.

Crsarmv7l, 2025-01-18:
> Hello! I test your file about 3-4 times with 10-15 repeats every tests - no one problem, exit every time on short press.

As I have said MULTIPLE TIMES it is intermittent, also why I provided a video. Just this morning i tried it 20 times without triggering the bug.

I am not going to go through this any more with you. You started off by telling me "long press for back is a feature"  even when I told you it isn't and there was no changelog stating that change was made. Now you said "well I tried it 3 times so there is no issue" despite a literal video.

While I find flipper's source too abstracted, I have contributed numerous features and fixes to both Radiolib, Rfcat and others and I can write C and C++ well enough to do what I want (if inefficiently). 

This bug will stay open because it is a bug with proof

Dmitry422, 2025-01-18:
Im not often use sub-ghz - my experience not so big. In some cases like barriers (locks) sub-ghz record exit only with long press, in another cases like your file exit by short press. So different cases - different presses.

I think in your cases may be hardware problem with key pad.

Crsarmv7l, 2025-01-18:
No. Long press is not a thing with the back button in subghz.

As I have previously mentioned, if it was the keypad it would happen in ALL
apps. It doesn't.

Crsarmv7l, 2025-01-18:
Additionally if this were a hardware issue, decoded subs that "stick" would exhibit the same "long press doesn't exit" they don't. A long press always exits a decoded sub.

 Raws hang for forever when they hang

Crsarmv7l, 2025-04-23:
This is much better in the latest release, but still hangs a bit occasionally. Not nearly as bad though

Dmitry422, 2026-01-01:
Not sure but as I know secure + v1 signal data not so clearly decoded now - so I think here can be some bugs.