DarkFlippers/unleashed-firmware #878: Vibro on power off screen

FalsePhilosopher, 2025-03-05:
### Description of the feature you're suggesting.

Please add a vibration to the power off scene, with a broken screen you can't tell if you are on the power off screen or not.

xMasterX, 2025-03-06:
It is present for a very long time, when you power off with holding back button and then pressing right arrow, it vibrates

closed 2025-03-06 by xMasterX

FalsePhilosopher, 2025-03-06:
> It is present for a very long time, when you power off with holding back button and then pressing right arrow, it vibrates

I'm not talking about the vibration confirming the power off, I'm talking about adding a vibration after the timed held back button at the power off request scene. Separate from the vibration to confirm the power off vibration to confirm you are on the power off request scene.