DarkFlippers/unleashed-firmware #879: subghz_transmitter_deserialize function bug

SpenserCai, 2025-03-06:
### Describe the bug.

maybe have a bug in subghz_transmitter_deserialize, i'm try send diff btn  with pt
first the last key bit is 03 then i changed it,before subghz_transmitter_deserialize is new,but than become old one,and i get data in hackrf,also get old btn code(03)
https://github.com/SpenserCai/flipperzero-subbrute/blob/master/helpers/subbrute_worker.c#L375

[image]

### Reproduction

like this code: https://github.com/SpenserCai/flipperzero-subbrute/blob/master/helpers/subbrute_worker.c#L375
this bug may be in firmware's sdk

### Logs

```Text

```

xMasterX, 2025-03-06:
Your issue present only on that specific princeton protocol code
at line 387 in `lib/subghz/protocols/princeton.c`
there is `flipper_format_update_hex` which updates the key in stream 
that is used for button replacement options in regular subghz app

If you use key with button code other than exisitng ones like 0x30
It would work fine

[image]
This code "destroys" the key in your case

xMasterX, 2025-03-06:
What it means that current princeton protocol implementation is not very compatible with subbrute app since it may modify key during the transmission

SpenserCai, 2025-03-07:
> What it means that current princeton protocol implementation is not very compatible with subbrute app since it may modify key during the transmission (after it)

I found that when you start brute for the first time after power-on, do not do any launch. Set the key code first. At this time, the key will not be changed. After launching once, it cannot be modified to any other key code, unless I restart the flipper or use a regular subghz app to launch the pt protocol data after saving the file.

Btw, is there any way to solve this problem temporarily?

Thanks so much

SpenserCai, 2025-03-07:
@xMasterX may be i find the way is subbrute to fix that：use subghz_custom_btn_reset

closed 2025-03-07 by SpenserCai