DarkFlippers/unleashed-firmware #892: Basic build fails with "Memory layout looks suspicious"

Niko-O, 2025-04-13:
### Describe the bug.

I am following the "How to Build by yourself" guide for Windows: https://github.com/DarkFlippers/unleashed-firmware/blob/dev/documentation/HowToBuild.md#how-to-build-by-yourself
Git is installed.
Cloning the Git repo seems to work.
`./fbt.cmd updater_package` does a ton of things and then outputs this:
```
2025-04-13 16:48:02,916 [WARNING] Firmware image overlaps C2 region and is not programmable!
2025-04-13 16:48:02,917 [WARNING] Memory layout looks suspicious
2025-04-13 16:48:02,917 [ERROR] You might brick your device into a state in which you'd need an SWD programmer to fix it.
2025-04-13 16:48:02,917 [ERROR] Please confirm that you REALLY want to do that with --I-understand-what-I-am-doing=yes
scons: *** [dist_updater_package] Error 2
```

Complete console log (too large to paste here):
[Log.txt](https://github.com/user-attachments/files/19726027/Log.txt)

### Reproduction

1. `git clone --recursive https://github.com/DarkFlippers/unleashed-firmware.git`
2. `cd .\unleashed-firmware\`
3. `./fbt.cmd updater_package`

### Target

I don't think it's relevant at this point, but if it is, please let me know what info exactly you need here.

### Logs

```Text
See attached Log.txt file.
```

### Anything else?

Note that I am basically completely new to programming for the Flipper Zero and there's a good chance that I'm missing something that seems obvious to you but isn't explained in the guide. For example, navigating the current working directory to the  directory created by `git clone` before running `fbt` is probably obvious to you but it is not not explained in the guide, so I had to guess whether or not I need to do that.

Niko-O, 2025-04-13:
1. Did I overlook a thing that explains this fact?
2. In any case, I recommend noting that in the build instructions, and updating them to specify commands that work by default.
3. Regarding `COMPACT=1 DEBUG=0`: Is `COMPACT=1 DEBUG=1` for debugging purposes valid?

xMasterX, 2025-04-13:
> 1. Did I overlook a thing that explains this fact?
> 2. In any case, I recommend noting that in the build instructions, and updating them to specify commands that work by default.
> 3. Regarding `COMPACT=1 DEBUG=0`: Is `COMPACT=1 DEBUG=1` for debugging purposes valid?

1 - no, its not stated in readme's right now
2 - sure, but it might be temporary thing, if we find a way to decrease fw size both build types will work
3 - incorrect, use only COMPACT=1 DEBUG=0, most thing affecting the debug is actually COMPACT = 0
but "release" type builds with COMPACT=1 are working with debug properly, it just might not show whats stored in some remote variables, if you use breakpoints or encounter a crash, you will see full stack calls tree just fine with compact=1

Niko-O, 2025-04-13:
Thank you for the explanation!
One more question: If I use `./fbt.cmd flash_usb`, do I also need to specify `COMPACT=1 DEBUG=0` there?

xMasterX, 2025-04-13:
> Thank you for the explanation! One more question: If I use `./fbt.cmd flash_usb`, do I also need to specify `COMPACT=1 DEBUG=0` there?

Yes you need to use `COMPACT=1 DEBUG=0` in that command too

Niko-O, 2025-04-14:
That worked, thanks!